mantidproject/mantid #31965: Workspace Calculator causes a crash when generating lots of workspaces

robertapplin, 2021-07-05:
### Expected behavior
Generating a large number of workspaces shouldn't cause a crash from the workspace calculator.

### Actual behavior
If I generate a large number of workspaces from a script or a GUI the workspace calculator causes a crash. It appears to happen during a resize event.

[image]

### Steps to reproduce the behavior
1. Open Workbench and run the script below.
2. While the script is running you can switch to the WorkspaceCalculator view and see that the embedded `WorkspaceSelector`'s keep resizing.
3. Eventually there will be a crash. Re-run the script if there isn't

```py
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

for i in range(500):
    CreateSampleWorkspace(OutputWorkspace="ws" + str(i))
```

### Platforms affected
Found on windows

robertapplin, 2021-07-05:
@darominski would you know how to fix this?

darominski, 2021-07-05:
I am looking into this. I can reproduce this issue on Ubuntu 18.04, albeit there the number of workspaces that are created before things go wrong is much larger, always above 8000. Then, there is a segfault in `QStandardItem *QStandardItem::child(int row, int column = 0) const`, but its origin seems to be the same.

`WorkspaceCalculator` uses `WorkspaceSelector` widget and it is always listening to ADS. It seems that there are more workspaces than a WorkspaceSelector can handle. I would like to test it against workspace selectors in different widgets as well. One way to address this issue would be to limit the maximal number of items the `WorkspaceSelector` holds, e.g. to 200-300. This approach seems to work fine in my tests with number of workspaces up to 20k. The cost is the limitation of functionality: the auto-completion and selecting of workspaces that are created when the counter is above the limit does not work (they are not added to the model, naturally), but when there are typed-in, it is possible to retrieve the desired workspace.

The other solution would be to ensure that when a widget is not visible from workbench, then it is not listening to any events. I must say am not sure how to approach it from this direction at the moment.

martyngigg, 2021-07-06:
> The other solution would be to ensure that when a widget is not visible from workbench, then it is not listening to any events. I must say am not sure how to approach it from this direction at the moment.

I think this would be preferable. Can we use either the show/hide/closeEvent methods to subscribe/unsubscribe to the events as needed?

PR #31973 (2021-07-08): WorkspaceSelector optional sorting
**Description of work.**

An issue coming from `WorkspaceCalculator` has been found when too many workspaces are created. This is caused by the time it takes to sort the model of the `WorkspaceSelector`. It seems that the next signal to add a workspace to the model comes when the sorting is not finished, thus creating issues when the number of workspaces is too large (the exact number is machine-dependent). The proposed solution is to put the sorting behind the parameter `m_sorted,` that can be set using `setSorted` method. By default, sorting is not enabled. [... cut]
closed 2021-07-08 by gvardany