mantidproject/mantid #31967: PSIBackgroundSubtraction - Allow passing in a custom function with different parameter values to default

robertapplin, 2021-07-05:
### Description
A couple of changes/additions are required for the `PSIBackgroundSubtraction` algorithm to allow background corrections from the Muon Analysis and FDA interfaces how we want them. Things to address:

- [ ] Currently, the default function is a FlatBack + Exp Muon Decay with no option to remove the Exp Muon Decay. We need to be able to only have a FlatBack. Need to consider backwards compatibility before making a change in this area.
- [ ] We need to be able to pass in a fit function (e.g. FlatBack) with parameter values that are not default. This is not possible at the moment.
- [ ] We need the chi squared as an output property of the algorithm.

AnthonyLim23, 2021-07-06:
I think it is possible that we could get rid of the algorithm and use a combination of fit and minus. 

The bit of functionality you will need to be careful to replicate is the estimates for the start and end x values.

AnthonyLim23, 2021-07-26:
@robertapplin should we close this?

robertapplin, 2021-07-26:
Yes I think so

closed 2021-07-26 by robertapplin