mantidproject/mantid #32022: Implement preprocessing for ISIS NR 2D detectors

gemmaguest, 2021-07-14:
**Original reporter:** Max at ISIS

A calibration step needs to be applied to NR data as a preprocessing step before we run our current reduction. An example for a linear detector is given in [this example](https://github.com/mantidproject/mantid/files/6817401/calibration.zip). This is very specific for a particular case with hard-coded spectrum numbers etc., and I'm unclear at the moment how to make this more generic (i.e. it should be made as a generic step as part of the ISIS NR workflow algorithm `ReflectometryISISLoadAndProcess`). We also need test data for the 2D detector case. 

This is a requirement as part of #31069 because this calibration (and possibly other preprocessing) should be done before we can start selection of regions of interest. 

We should create a new algorithm to perform preprocessing of ISIS NR runs and call this as part of `ReflectometryISISLoadAndProcess`. It also needs to be callable as a standalone algorithm for example as part of #31069 we will want to perform preprocessing before we select regions of interest, and then continue the rest of the processing after we have the ROIs. `ReflectometryISISLoadAndProcess` may need some refactoring to make this easier.

gemmaguest, 2022-06-20:
Max is putting together a new example for this. The original example needed to know which pixel was the specular, but he thinks there should be enough info in the IDF which should place the specular pixel at 0 if theta is 0

rbauststfc, 2022-09-06:
We've done some investigating and think we can find the specular pixel by taking the 2 theta value from the sample log file and finding the detector pixel with the closest 2 theta value. Code for doing this is given in the attached amended version of the example calibration script:

[Calibration_test - Copy.zip](https://github.com/mantidproject/mantid/files/9498272/Calibration_test.-.Copy.zip)

The plan is to convert this script into a new Reflectometry algorithm that can eventually be called from `ReflectometryISISLoadAndProcess`. The `.dat` file name will need to be passed as a parameter.

rbauststfc, 2022-09-08:
The example script provided defines a spectra variable that gives the workspace indices of the detectors that need to be calibrated. The range values have been hard-coded into the script. We think that the calibration should be applied to all detectors in the current component (i.e. the top level component parent of the specular pixel), which should be possible to find programmatically. We need to exclude any monitors and non-detector components.

Only the detectors in the current component are included so that we don't pick up any unused detectors that are still defined in the IDF.

closed 2022-11-18 by martyngigg