mantidproject/mantid #32032: Investigate hard crash in workbench

gvardany, 2021-07-16:
Workbench crashes sporadically, especially when running a script that generates a lot of workspaces and groups in ADS.
Most likely, it's another race condition in workspace tree widget, or a regression introduced recently (such as the workspace calculator).

### Expected behavior
No hard crash.

### Actual behavior
Crashes frequently, but sporadically.

### Steps to reproduce the behavior
This is not systematically reproducible, unfortunately.

### Platforms affected
Observed on macOS, might be macOS-only issue.

### Crash trace
```
Python(24955,0x10a02be00) malloc: *** error for object 0x7fc39bdf52c0: pointer being freed was not allocated Python(24955,0x10a02be00) malloc: *** set a breakpoint in malloc_error_break to debug

This is the full trace:

Process:               Python [25016]
Path:                  /usr/local/Cellar/python@3.9/3.9.5/Frameworks/Python.framework/Versions/3.9/Resources/Python.app/Contents/MacOS/Python
Identifier:            org.python.python
Version:               3.9.5 (3.9.5)
Code Type:             X86-64 (Native)
Parent Process:        MantidWorkbench [25015]
Responsible:           Terminal [2500]
User ID:               103532

Date/Time:             2021-07-12 12:10:17.461 +0200
OS Version:            macOS 11.4 (20F71)
Report Version:        12
Anonymous UUID:        E55A4C87-6D73-E62D-EDF8-D8ABCABA6D5F

Sleep/Wake UUID:       83CC6A2E-361E-46B2-955A-926DD06AE793

Time Awake Since Boot: 210000 seconds
Time Since Wake:       3700 seconds

System Integrity Protection: disabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [25016]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.qt-project.QtGui          	0x0000000133521327 QStandardItemModel::data(QModelIndex const&, int) const + 119
1   org.qt-project.QtWidgets      	0x000000013405a037 0x133cc3000 + 3764279
2   org.qt-project.QtWidgets      	0x000000013405a768 0x133cc3000 + 3766120
3   org.qt-project.QtWidgets      	0x0000000134057a7d 0x133cc3000 + 3754621
4   org.qt-project.QtWidgets      	0x000000013405754a 0x133cc3000 + 3753290
5   org.qt-project.QtCore         	0x0000000128987614 QObject::event(QEvent*) + 900
6   org.qt-project.QtWidgets      	0x0000000133cd6a56 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 262
7   org.qt-project.QtWidgets      	0x0000000133cd7e02 QApplication::notify(QObject*, QEvent*) + 466
8   QtWidgets.abi3.so             	0x000000013397319a sipQApplication::notify(QObject*, QEvent*) + 122
9   org.qt-project.QtCore         	0x000000012895e2b6 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 166
10  org.qt-project.QtCore         	0x000000012895f3f3 QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 803
11  libqcocoa.dylib               	0x000000014206e892 0x142032000 + 247954
12  libqcocoa.dylib               	0x000000014206ef98 0x142032000 + 249752
13  com.apple.CoreFoundation      	0x00007fff2041e6dc __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
14  com.apple.CoreFoundation      	0x00007fff2041e644 __CFRunLoopDoSource0 + 180
15  com.apple.CoreFoundation      	0x00007fff2041e3ba __CFRunLoopDoSources0 + 242
16  com.apple.CoreFoundation      	0x00007fff2041cdc8 __CFRunLoopRun + 897
17  com.apple.CoreFoundation      	0x00007fff2041c380 CFRunLoopRunSpecific + 567
18  com.apple.HIToolbox           	0x00007fff288bcab3 RunCurrentEventLoopInMode + 292
19  com.apple.HIToolbox           	0x00007fff288bc815 ReceiveNextEventCommon + 587
20  com.apple.HIToolbox           	0x00007fff288bc5b3 _BlockUntilNextEventMatchingListInModeWithFilter + 70
21  com.apple.AppKit              	0x00007fff22b476f2 _DPSNextEvent + 864
22  com.apple.AppKit              	0x00007fff22b45ec5 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1364
23  com.apple.AppKit              	0x00007fff22b38239 -[NSApplication run] + 586
24  libqcocoa.dylib               	0x000000014206dd74 0x142032000 + 245108
25  org.qt-project.QtCore         	0x000000012895a6b7 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 471
26  org.qt-project.QtCore         	0x000000012895e852 QCoreApplication::exec() + 130
27  QtWidgets.abi3.so             	0x00000001339740d0 meth_QApplication_exec_(_object*, _object*) + 80
28  org.python.python             	0x000000010bb970c2 cfunction_call + 90
29  org.python.python             	0x000000010bb4c6da _PyObject_MakeTpCall + 129
30  org.python.python             	0x000000010bc44cbd call_function + 270
31  org.python.python             	0x000000010bc42526 _PyEval_EvalFrameDefault + 45324
32  org.python.python             	0x000000010bb4d240 _PyFunction_Vectorcall + 192
33  org.python.python             	0x000000010bc44c53 call_function + 164
34  org.python.python             	0x000000010bc42562 _PyEval_EvalFrameDefault + 45384
35  org.python.python             	0x000000010bb4d240 _PyFunction_Vectorcall + 192
36  org.python.python             	0x000000010bc44c53 call_function + 164
37  org.python.python             	0x000000010bc42562 _PyEval_EvalFrameDefault + 45384
38  org.python.python             	0x000000010bb4d240 _PyFunction_Vectorcall + 192
39  org.python.python             	0x000000010bc44c53 call_function + 164
40  org.python.python             	0x000000010bc42562 _PyEval_EvalFrameDefault + 45384
41  org.python.python             	0x000000010bb4d240 _PyFunction_Vectorcall + 192
42  org.python.python             	0x000000010bc44c53 call_function + 164
43  org.python.python             	0x000000010bc42562 _PyEval_EvalFrameDefault + 45384
44  org.python.python             	0x000000010bc362d6 _PyEval_EvalCode + 403
45  org.python.python             	0x000000010bc3104f builtin_exec + 389
46  org.python.python             	0x000000010bb97a43 cfunction_vectorcall_FASTCALL + 95
47  org.python.python             	0x000000010bc44c53 call_function + 164
48  org.python.python             	0x000000010bc42562 _PyEval_EvalFrameDefault + 45384
49  org.python.python             	0x000000010bc362d6 _PyEval_EvalCode + 403
50  org.python.python             	0x000000010bb4d2f8 _PyFunction_Vectorcall + 376
51  org.python.python             	0x000000010bc44c53 call_function + 164
52  org.python.python             	0x000000010bc42562 _PyEval_EvalFrameDefault + 45384
53  org.python.python             	0x000000010bc362d6 _PyEval_EvalCode + 403
54  org.python.python             	0x000000010bb4d2f8 _PyFunction_Vectorcall + 376
55  org.python.python             	0x000000010bcb4fd0 pymain_run_module + 212
56  org.python.python             	0x000000010bcb4a9a pymain_run_python + 433
57  org.python.python             	0x000000010bcb48ad Py_RunMain + 23
58  org.python.python             	0x000000010bcb5bca pymain_main + 35
59  org.python.python             	0x000000010bcb5ea0 Py_BytesMain + 42
60  libdyld.dylib                 	0x00007fff20340f5d start + 1
```

darominski, 2021-07-17:
I am able to consistently reproduce the issue with the following script:

```
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

ws_list = []

for i in range(50000):
    print(i)
    name = "ws" + str(i)
    CreateSingleValuedWorkspace(OutputWorkspace=name, DataValue=0)
    ws_list.append(name)
    if i % 10 == 0:
        DeleteWorkspaces(WorkspaceList=ws_list)
        ws_list.clear()
```

The issue does not occur when the workspaces are only created, to properly have a consistent segmentation violation one needs to also remove them from the ADS. 

I did some testing and removed the WorkspaceCalculator from the workbench layout completely, and the issue did occur. Thus, I believe it is linked with the `WorkspaceSelector` always listening to the ADS changes via the `WorkspaceCalculator`. I also believe the proper solution will involve making `WorkspaceSelector` desactivated or disconnected from the signals when not in use.

gvardany, 2021-07-26:
Thanks for investigating this further. Indeed, it was happening when deleting workspaces and yes this is related to the workspace selector in the calculator. This makes it a high priority as it will cause crashes even if the calculator is hidden. I think it makes sense if the selectors do not observe ADS, until one clicks on them. Still, this will only fix the symptom. While we have a reproducible crash, we should aim to fix also the source, which is most likely a race condition or a logic error in the workspace selector widget's model.

gvardany, 2021-08-25:
Closed with https://github.com/mantidproject/mantid/pull/32076

closed 2021-08-25 by gvardany