mantidproject/mantid #32065: WinError 145 on closing Mantid

sf1919, 2021-07-22:
**Describe the bug**
On closing Mantid the following error is sometimes occuring

```
 File "C:\MantidNightlyInstall\bin\workbench\app\mainwindow.py", line 588, in closeEvent
   self.project_recovery.remove_current_pid_folder()
 File "C:\MantidNightlyInstall\bin\workbench\projectrecovery\projectrecovery.py", line 199, in remove_current_pid_folder
   self._remove_directory_and_directory_trees(self.recovery_directory_pid)
 File "C:\MantidNightlyInstall\bin\workbench\projectrecovery\projectrecovery.py", line 138, in _remove_directory_and_directory_trees
   shutil.rmtree(directory)
 File "C:\MantidNightlyInstall\bin\lib\shutil.py", line 730, in rmtree
   return _rmtree_unsafe(path, onerror)
 File "C:\MantidNightlyInstall\bin\lib\shutil.py", line 603, in _rmtree_unsafe
   _rmtree_unsafe(fullname, onerror)
 File "C:\MantidNightlyInstall\bin\lib\shutil.py", line 612, in _rmtree_unsafe
   onerror(os.rmdir, path, sys.exc_info())
 File "C:\MantidNightlyInstall\bin\lib\shutil.py", line 610, in _rmtree_unsafe
   os.rmdir(path)
OSError: [WinError 145] The directory is not empty: '...\mantidproject\\mantid\\workbench-recovery\....'
```

**To Reproduce**
The issue occurs when closing Mantid but not everytime and is therefore difficult to reproduce.

**Expected behavior**
When closing Mantid there should be no crashes or errors.

**Platform/Version (please complete the following information):**
Mantid version 5.0.20200909.1539 
Mantid version 6.1.20210719.2059

Windows

sf1919, 2021-07-22:
I've been provided with some additional information. Aidy loaded an old project then left it open for a day or so before closing. He also had multiple instances of mantid open at once.

AnthonyLim23, 2021-07-26:
I get this a lot. I only have one version of mantid open, I think its related to it trying to close and record project recovery at the same time.

The big problem is that it locks all of my dll's so I have to restart my machine to be able to rebuild Mantid

robertapplin, 2021-08-25:
This issue was again found during manual testing (see #32306). It was hard to reproduce

DavidFair, 2021-08-25:
Could be similar to: https://bugs.python.org/issue33240
Do we have any nested folders that are being held open somewhere? Or a race between handles I wonder :thinking: 

One way to easily tell is trying to delete the tree whilst Mantid is open, if it fails because something is holding a handle it's more likely we're racing that

sf1919, 2021-11-16:
This error also occurred when a user moved some files to a local directory. When they browsed to one of these files and loaded it, it appears to load and make the workspace correctly but reports ‘no data loaded’ and doesn’t extract the times and run information to show on the home tab. It also produced an Mantid error message citing [WinError 145].

martyngigg, 2022-01-17:
I had a quick peek at the code around here. My guess at what could be happening based on a recent comment from a user that it seemed to happend after a "large" reduction.

1) In the workbench [`closeEvent`](https://github.com/mantidproject/mantid/blob/main/qt/applications/workbench/workbench/app/mainwindow.py#L601) we stop the recovery saving thread
2) we then immediately try to [remove the pid directory](https://github.com/mantidproject/mantid/blob/main/qt/applications/workbench/workbench/app/mainwindow.py#L603) to clean up the saved files

I suspect that for large reductions, which could means many files rather than 1 big one, then the recovery has been asked to stop but has not yet finished and is still writing to that pid directory. When we then try to remove the directory it falls over as we don't [handle any exceptions](https://github.com/mantidproject/mantid/blob/main/qt/applications/workbench/workbench/projectrecovery/projectrecovery.py#L195).

AnthonyLim23, 2022-01-17:
That would make sense. For muons the files themselves are not large. However, we have lots of them

martyngigg, 2022-01-17:
> That would make sense. For muons the files themselves are not large. However, we have lots of them

Thanks for confirmin. The user I mentioned is from SANS and they tend to have lots of workspacs too.

martyngigg, 2022-01-21:
Fixed by #33335

PR #33335 (2022-01-20): Handle OSError exceptions when trying to remove recovery checkpoints
**Description of work.**

We now trap errors that would have caused the error reporter to be raised. A user can do nothing about
the errors and recovery cleans up after itself in future runs so the risk is minimal. See the commit messages
for more details.

**To test:**

* Please pay close attention to code changes amd review them throughly to check the logic.
* Run through the [project recovery manual tests](https://developer.mantidproject.org/Testing/ErrorReporter-ProjectRecovery/ProjectRecoveryTesting.html)

Fixes #32065 
 [... cut]
closed 2022-01-21 by martyngigg