mantidproject/mantid #32115: Add cut along arbitrary direction (not axis aligned) in sliceviewer

RichardWaiteSTFC, 2021-08-02:
**Is your feature request related to a problem? Please describe.**
MantidPlot had a feature that allowed for a cut to be made along an arbitrary direction in the plotted plane.

**Describe the solution you'd like**
Perhaps for the box feature to include an  orientation?
MantidPlot opened a separate plot on the RHS but it was always a bit busy.

**Describe alternatives you've considered**
Presently the WISH scientists use BinMD if they need to make a cut along a direction that isn't aligned to the plotting axes - but this is more time consuming and prone to mistakes (wrong projection axes defined etc.)

**Additional context**

zjmorgan, 2021-08-16:
@RichardWaiteSTFC This is the proposed criteria for MDEvent. Does this make much sense for your side?

```
For the non-axis aligned line cut:
1. Create/load MD Event workspace 
2. Click on linecut tool
3. Draw a line by clicking (region of interest)
 - Click for one end and click to select the other end
 - The end points should be selectable can be dragged to anywhere on the current slice
4. A preview of the line cut should open up
 - It can open to the right hand side 
 - It should close any open peaks table viewers
5. There should be a way to drag the line around as a rigid line
   - In MantidPlot the line can be dragged around but this is not the case in MantidWorkbench
   - One possibility is to select the center of the line and translate it around

For the line viewer
1. The end points should be editable 
2. The integration thickness of the remaining axes should be editable
3. Option to set the number of bins or bin width along the line.
4. Option to plot semilog-y axes or normal axes
5. Option to select the horizontal axes of the line viewer
  - Option AxisX
  - Option AxisY
  - Option sqrt(AxisX^2+AxisY^2) 
6. Option to apply no normalization or normalization by volume or number of events 
```

RichardWaiteSTFC, 2021-08-16:
Thanks @zjmorgan, I think we should discuss this with @DannyHindson and @martyngigg (who is on holiday till 19th Aug)  as well, but my initial thoughts are

RE non-axis aligned cut
3. Agreed from UX perspective the line cut should be done by clicking start and end-point rather than drawing/rotating a rectangle.
4. If the plot is on RHS then agreed we have to close peaks viewer (no requirement to look at both simultaneously) - one advantage of displaying on RHS is plenty of room for options such as integration width, nbins etc.
**6. I would add that the integration width should be visually represented on the line cut on the 2D image - I believe this was also done in MantidPlot**

RE line viewer
5. Not clear what is meant here - is it that having selected a ROI in Q space you want option to plot along axis of cut but also give option to choose x-axis to be the orthogonal directions or |Q|?
7. Could use normalisation in color bar to avoid an extra box or do you think this would be non-intuitive?

Perhaps we should also discuss whether we need a line viewer plot inside the sliceviewer. If the line cut tool creates a 1D MD object (output of `BinMD`) this can be plotted externally in a matplotlib figure window. This has the advantage of giving the user access to the nice tools to change line properties etc. and might help with space constraints. Such a plot would automatically be updated if the cut were changed (ADS observer already does this). Just a thing to consider - I don't feel very strongly about it!

zjmorgan, 2021-08-17:
> I would add that the integration width should be visually represented on the line cut on the 2D image - I believe this was also done in MantidPlot  

I agree with that. That is really helpful.

> Not clear what is meant here - is it that having selected a ROI in Q space you want option to plot along axis of cut but also give option to choose x-axis to be the orthogonal directions or |Q|?

No, I meant that the line cut can be dragged around with fixed orientation and length. This would happen if you clicked on the center. This isn't a hard requirement but I believe users would find that useful.

>  Could use normalisation in color bar to avoid an extra box or do you think this would be non-intuitive?

I suppose users may want to change the 1d plot normalization directly without affecting the 2d slice plot. But that's more of my opinion. I think it would be acceptable either way.

zjmorgan, 2021-08-17:
> Perhaps we should also discuss whether we need a line viewer plot inside the sliceviewer. If the line cut tool creates a 1D MD object (output of BinMD) this can be plotted externally in a matplotlib [...]

I am also okay with this. I think this also a good solution if space is a concern in the slice viewer. I think the main thing we would want to ensure is that the 1d plot automatically open so user's know it exists.

RichardWaiteSTFC, 2021-08-17:
Thanks, I take your point about being able to change the two independently and the widget having it's own box for the normalisation might be easier to code as well. Also agree about being able to drag the cut around.

> No, I meant that the line cut can be dragged around with fixed orientation and length. This would happen if you clicked on the center. This isn't a hard requirement but I believe users would find [...]

But I think we're talking at cross-purposes on this point! My question referred to this requirement RE the line viewer

> 5. Option to select the horizontal axes of the line viewer
>   - Option AxisX
>   - Option AxisY
>   - Option sqrt(AxisX^2+AxisY^2)

zjmorgan, 2021-08-17:
> Option to select the horizontal axes of the line viewer
> - Option AxisX
>  - Option AxisY
> - Option sqrt(AxisX^2+AxisY^2) 
 
 What I meant here is for the horizontal axes of the 1d line plot. We can choose to plot directly along the cut `sqrt(AxisX^2+AxisY^2)` where `AxisX` and `AxisY` refer to the chosen axes of the slice viewer. But we could also just plot the projection of either `AxisX` or `AxisY` just like it does now with the current ROI plot. This isn't a hard requirement, but MantidPlot could do this.

DannyHindson, 2022-02-14:
We'd like to keep this issue open

DannyHindson, 2022-08-31:
Closing - feature was implemented on linked pull request

closed 2022-08-31 by DannyHindson