mantidproject/mantid #32189: EngDiff Focus plots different to docs

DanielMurphy22, 2021-08-11:
**Describe the bug**
The focus tab outputs plots that look incorrect, or at least different to the manual testing docs.

**To Reproduce**

- If it hasn't yet been merged, cheeckout this bugfix PR #32169
- Open Engineering Diffraction interface, and select Create New Calibration
- For the vanadium number enter 307521, and for the Calibration Sample number enter 305738, and click Calibrate.
- Then click on the focus tab and enter run number 305761.
- Notice that the output plots are different to the testing docs: https://developer.mantidproject.org/Testing/EngineeringDiffraction/EngineeringDiffractionTestGuide.html#focus

**Expected behavior**
I think the focussed plots should still look as in the manual testing docs. If it's correct, then the testing docs should be updated!

**Screenshots**
On the left is the manual testing docs and the right is the current output plots.
[image]

**Platform/Version:**
 - OS: found on Ubuntu 18.04
 - Mantid Version Using PR  #32169 to work around a bug

RichardWaiteSTFC, 2021-08-19:
The focussed data produced now (RHS image above) are as expected - it's because the data in the old screenshots were not normalised by vanadium correctly. You expect noisy and large intensities at low TOF - the spectra are divided by the vanadium spectra which in the region of low TOF (where there is little flux) have bins of small but non-zero intensity, leading to what looks like diverging intensity as TOF->0. However there are a couple of things we should do:
1. Replace the images in the docs
2. Crop the workspace to remove the noisy signal at low TOF (probably best to define these limits in lambda)

DanielMurphy22, 2021-08-23:
It looks like step 1 may have been handled here: https://github.com/mantidproject/mantid/pull/32252 but I will check

DanielMurphy22, 2021-08-23:
As mentioned above, the Manual testing docs have been updated, although I'll need to check with @RichardWaiteSTFC that they every detail is correct, but there is no need for this high priority issue. The second point about cropping is something we can discuss to raise an issue for next release.

closed 2021-08-23 by DanielMurphy22