mantidproject/mantid #32263: Manual Testing Documentation

martyngigg, 2021-08-20:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

In the build directory run `cmake docs-html` 
Check in the `docs/html` directoy:

* Algorithm, fit, concept and api pages should be generated
* Algorithm dialog snapshots should appear on algorithm pages in offline help
* Math formulae should appear on algorithm pages in offline help
* workflow diagrams should appear on algorithm pages in offline help

You may wish to use this script: [OpenMostDocumentationForTesting.py](https://github.com/mantidproject/mantid/blob/master/tools/scripts/OpenMostDocumentationForTesting.py) to open all the online docs pages!

martyngigg, 2021-09-01:
It looks like I accidentally deleted it when removing old tools that looked like they weren't used and written for Python 2. The last version of the file is at https://github.com/mantidproject/mantid/blob/2919bcd126d5bf96672da0bd1f12967e12d1da5a/tools/scripts/OpenMostDocumentationForTesting.py

DanielMurphy22, 2021-09-01:
Tested on Ubuntu 20.04. Nothing major found. We should remove the email address from https://docs.mantidproject.org/nightly/algorithms/Abins-v1.html

and the  auto-generated images on these 2 pages don't look great: 
- https://docs.mantidproject.org/nightly/interfaces/indirect/Indirect%20Elwin%20&%20Iqt.html
- https://docs.mantidproject.org/nightly/interfaces/indirect/Inelastic%20Data%20Analysis.html

closed 2021-09-01 by DanielMurphy22