mantidproject/mantid #32264: Manual Testing Workbench Group 1

martyngigg, 2021-08-20:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

https://developer.mantidproject.org/Testing/Core/Core.html

SilkeSchomann, 2021-08-24:
I've finished testing on IdAaaS. The only issue I found so far is that superplots are not included when saving a project.

DanielMurphy22, 2021-08-24:
@DavidFair also saw this here: https://github.com/mantidproject/mantid/issues/32265#issuecomment-903871436

Can I ask one of you to write an issue and label it with `Team: ILL`

DavidFair, 2021-08-24:
#32302

SilkeSchomann, 2021-08-25:
Today I am testing on windows with the nightly build from 24th of August

SilkeSchomann, 2021-08-25:
Unlike the IDAaaS version there is an error message:
[image]

SilkeSchomann, 2021-08-25:
The show and hide buttons on top of the plots toolbox do not seem to work. Clicking show/hide for an individual plot in the list does work though.

closed 2021-08-25 by SilkeSchomann