mantidproject/mantid #32265: Manual Testing Workbench Group 2

martyngigg, 2021-08-20:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

https://developer.mantidproject.org/Testing/Core/Core.html

DavidFair, 2021-08-23:
Testing on OSX:
- Superplot is not saved in Project Save / Load
- The help icon has been crushed in algorithm dialogs:
[image]
- Deprecation warnings with zooming and unzooming Color Fill plots on a workspace from CreateSampleWorkspace:
`Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.9 it will stop working`
- I also got another deprecation warning for numpy in slice-viewer, but I cannot reproduce it.
- I managed to manually dig down to IPython / Notebooks, both work, but we don't have icons / launchers despite shipping these on OSX?

DavidFair, 2021-08-24:
I'm going to test on Windows today....

DavidFair, 2021-08-25:
I'm going to close this as #32267 covered IDaaS

closed 2021-08-25 by DavidFair