mantidproject/mantid #32268: Manual Testing Muon

martyngigg, 2021-08-20:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See instructions at https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/index.html
https://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html

AnthonyLim23, 2021-08-25:
In elemental analysis, the remove subplot box is not automatically closed if not plots are left. I doubt a user will ever see this and it has been present for some time

AnthonyLim23, 2021-08-25:
In elemental analysis I can cause a crash https://github.com/mantidproject/mantid/issues/32309

AnthonyLim23, 2021-08-25:
ALC seems fine

gemmaguest, 2021-08-25:
I'll collate notes in this comment as I go.

- The instructions state that the grouping tab should be disabled when no data is loaded but it is not. I'm not sure if that's a bug or just confusing instructions

AnthonyLim23, 2021-08-25:
Phasequad does not work and instead throws an error

AnthonyLim23, 2021-08-25:
I think I know whats happened with tests 2 and 5 of group 2. 

In test 2 you need to update the first row in the table to have A=0.2 and phi=1.5 otherwise it gives rubbish
In test 5, you have the tiled by option set to Run instead of group/pair

gemmaguest, 2021-08-25:
> For "In Group 2 (fitting), Test1, my plot looks a little different:" did the values look right (8.5 and 0.4)?

I don't know I'm afraid

gemmaguest, 2021-08-25:
I'll start a new comment seeing as you already responded to the previous one :)

- In Group 2 (fitting), Test 8. The flat background parameter is 0.02. I'm not sure if this is the expected "closer to zero" result. I can't remember what it was originally but I think a small negative value.

AnthonyLim23, 2021-08-25:
> I'll start a new comment seeing as you already responded to the previous one :)
> 
> * In Group 2 (fitting), Test 8. The flat background parameter is 0.02. I'm not sure if this is the expected "closer to zero" result. I can't remember what it was originally but I think a small negative value.

Closer to zero means that the error is bigger than the value (so we accept a null hypothesis of 0 for the value)

closed 2021-09-09 by martyngigg