mantidproject/mantid #32269: Manual Testing TOFConverter

martyngigg, 2021-08-20:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

Make sure that inputs and outputs work sensibly, stress test with some bad inputs (e.g. letters in a numeric input)
See https://docs.mantidproject.org/interfaces/TOF%20Converter.html

sf1919, 2021-08-24:
I've attempted to test TOFConverter.

Firstly the documentation link does not work. This is the correct link - https://docs.mantidproject.org/nightly/interfaces/utility/TOF%20Converter.html

I tested the TOFConverter on Windows 10 using Nightly version 6.1.20210819. The input boxes only accept numbers and negative numbers - any attempt to type symbols or letters does not do anything which is great. If a 0 or a negaitve number is used a popup appears explaining the value is too small and a value greater than 0 must be entered. No crashes were observed. It was difficult to know what numbers to use and if the outputs were sensible/expected. Is there any further guidance for this, maybe some example numbers to use?

I also tested the TOFConverter on IDAaaS using Nightly version 6.1.20210820. The same was observed as with the Windows 10 tests. No crashes.

martyngigg, 2021-08-24:
I think the documentation link is worth fixing for this release. Can you open an issue, leave it unassigned, and put it in the Release 6.2 milestone? It'll be in the `New Issues` column for someone to pick up then.

DanielMurphy22, 2021-08-24:
Is the documentation link the help button on the interface, or the link to test instructions in this issue? If its the latter, then I can update the spreadsheet of Manual test instructions.

martyngigg, 2021-08-24:
> Is the documentation link the help button on the interface, or the link to test instructions in this issue? If its the latter, then I can update the spreadsheet of Manual test instructions.

Ah yes I think you're right 🤦🏻

martyngigg, 2021-08-24:
> I tested the TOFConverter on Windows 10 using Nightly version 6.1.20210819. The input boxes only accept numbers and negative numbers - any attempt to type symbols or letters does not do anything [...]

It sounds like we might need some acceptance testing pages for the TofConverter?

sf1919, 2021-08-24:
It's definitely the link in the instructions that were incorrect. I will double check the link in Mantid itself is correct also.

sf1919, 2021-08-24:
Yeah link in Mantid itself is correct so its just the testing instructions that need updating. I don't think that will need a new issue, will it @martyngigg ?

martyngigg, 2021-08-24:
Thanks for checking. No, no need for an issue there.

sf1919, 2021-08-25:
Am I ok to close this testing? Or will there be an acceptance test page for me to use for the calculations? As stated above I have no idea if I'm getting sensible results from the numbers I'm inputting but otherwise it works fine on both Windows and IDAaaS. @martyngigg @DanielMurphy22

DanielMurphy22, 2021-08-25:
Yes please close and I'll make sure to correct the link!

closed 2021-08-25 by sf1919