mantidproject/mantid #32271: Manual Testing Engineering Diffraction

martyngigg, 2021-08-20:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

Follow: https://developer.mantidproject.org/Testing/EngineeringDiffraction/EngineeringDiffractionTestGuide.html

DannyHindson, 2021-08-20:
We have just opened a PR to update the testing guide for the engineering diffraction so best to wait until Monday to start this testing - in case you were thinking of starting today!

RichardWaiteSTFC, 2021-09-01:
> Test 1 Focus: The tests say that you should get a multiplot with both banks, instead following these steps you only get one plot containing bank 2. Trying without doing the Load existing calibration in a new RB number plots both banks.
> Also the plots in the manual testing look like they have been rebinned.
> 
> Test 3: this mentions Crop focus, this is not an option available, it looks like it just gets it from calibration.
> 
> Test 4: I note that the Load browser on the fitting is a different type to the others? just curious as to why.
> When you have undocked the plot the remaining plot function buttons remain in an area with a minimum size too large to it, I don't know if there is a better way for the gui to function but it just looks awkward.
> 
> Test 5: IMPORTANT go to fitting, set add to plot to true, load a focussed bank, load a second, there will now be 3 coppies of the first bank along with one of the second.
> 
> [image]
> 
> test 6 if you enter values in Niter or XWindow that are too small you get a sad mantid.

RE Test 4 The qt file browsing widget must be used instead of the native OS one if filters are to be applied to the filenames (in this case TOF/dSpacing and bank numbers). Wrt the plot function buttonns on the toolbar, this doesn't always happen but it depends on the size of your window and display etc. and whether you have the fit browser visible  - the window can always be made larger to make all the options visible. 

I will make issues for the rest (update docs for Test 1 and 3) and look into bugs related to Test 5 and 6

closed 2021-09-02 by RichardWaiteSTFC