mantidproject/mantid #32274: Manual Testing ISIS Reflectometry

martyngigg, 2021-08-20:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

For testing procedures see http://developer.mantidproject.org/Testing/ReflectometryGUI/ReflectometryGUITests.html

gemmaguest, 2021-09-01:
> 
> 
> In the Transferring Runs section it all works as expected, except runs that are highlighted in blue are not transferred.
> 
> @gemmaguest What does a run being highlighted in blue mean? Are blue runs bad or not able to be transferred?
> 
> [image]

Yes, exactly that - blue means error (i.e. cannot be transferred) and if you hover over a blue row it will give a tooltip explaining the problem. I know the blue highlighting is not particularly intuitive but it's easy for users once they get used to it.

gemmaguest, 2021-09-01:
> 
> 
> Just a small usability thing. If I process all then groups with more than one run have their title highlighted in green, but groups with only one run do not get their title highlighted in green [image]

Again this is intentional - when a group has multiple rows it requires postprocessing and it turns green when postprocessing is complete. If there is only one row then postprocessing is not applicable so it doesn't have state per se. (Although I'd quite like to highlight it a different colour to indicate all the child rows are complete, but it's very low priority.)

gemmaguest, 2021-09-01:
> 
> 
> All looks fine, tested on windows. I just have two usability thing with the table:
> 
>     1. Pressing enter on a new value will create a new row (maybe not all the time). I find this slightly unintuitive as pressing enter to confirm a value feels the most natural way of doing things, but then again I'm not a user of this interface.

I think the users are happy with the behaviour in general - they often have lots of runs to enter and like the behaviour to be more like an excel spreadsheet where pressing enter will move to the next row. I agree it is not always intuitive though especially to "finalise" data entry. It would be nice to improve but not something that bothers them so is very low priority.

>     2. Also, these new blank rows interfere with the progress bar, e.g if we had 2 groups each with 2 rows and add 1 blank row the progress bar caps at 83% if we do a "process all".
 
Ah that sounds like a bit of a bug. It's very low priority though and not something we'd likely to get round to fixing - this would be a small effect when there are a lot of runs, and I think the the progress bar is just used as a rough indication anyway.

DanielMurphy22, 2021-09-01:
@gemmaguest Thanks for clarifying our questions. Nothing worth fixing seems to have been found so I'll close this now.

closed 2021-09-01 by DanielMurphy22