mantidproject/mantid #32275: Manual Testing ISIS SANS (new GUI)

martyngigg, 2021-08-20:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See https://developer.mantidproject.org/Testing/SANSGUI/SANSGUITests.html

DavidFair, 2021-08-23:
@AnthonyLim23 Just an FYI: You'll need to download the latest ISIS Usage data, since the files within have updated this release

AnthonyLim23, 2021-08-23:
just downloaded the training data this morning

AnthonyLim23, 2021-08-25:
I cant see the full names of the different sections (e.g. Beam Cent)

AnthonyLim23, 2021-08-25:
Step 5 is unclear as I closed manage user directories before setting the save path (as that is what it suggests)

AnthonyLim23, 2021-08-25:
when I have loaded a batch file, the process and "load" button are enabled, but I dont have any data to use them on. Is this intended?

AnthonyLim23, 2021-08-25:
I cannot use tab to move across the cells in a row

AnthonyLim23, 2021-08-25:
I changed the z offset from -8 to 8 and then edited the table. The value had reset to -8

AnthonyLim23, 2021-08-25:
I got the load of the batch file to fail with the attached (might be the blank cols or rows)
[test.csv](https://github.com/mantidproject/mantid/files/7045399/test.csv)

AnthonyLim23, 2021-08-25:
I can enter letters in to any of the cols

AnthonyLim23, 2021-08-25:
both the settings and beam centre tabs reset after loading the user file

AnthonyLim23, 2021-08-25:
I seem unable to load the `txt` file

AnthonyLim23, 2021-08-25:
@DavidFair Im not sure what I have done, but I cannot enter any user files anymore. Even after restarting mantid. Going to try an uninstall and reinstall.

AnthonyLim23, 2021-08-25:
The Q, Wavelength, Detector Limits sub-tab is almost impossible to read (Im using a 4k monitor) 
[image]

AnthonyLim23, 2021-08-25:
Step 5 in the mask sections doesnt match the GUI. It says to change the 0, but what I can see if from -90 to 90

AnthonyLim23, 2021-08-25:
I get 8 output files instead of 12 (6 per row, 2 rows)

AnthonyLim23, 2021-08-25:
After selecting 2D for reduction, CanSAS(1D) is still selected and enabled

AnthonyLim23, 2021-08-25:
for merged reduction, I get 8 lines and not 6 (but for each row there is a repeat- HAB)

AnthonyLim23, 2021-08-25:
I cant see the checkboxes for this step ```Both the Update main-detector and Update Hab checkboxes should be checked and enabled.
```

AnthonyLim23, 2021-08-25:
I used the default set up for the beam centre finder, clicked run. When the plot appeared I looked at it then closed it. That caused a hang and Python died

AnthonyLim23, 2021-08-25:
For sum runs I cannot see a Select Save Directory option

AnthonyLim23, 2021-08-25:
In the Diagnostic tab, I can set the Period to 5 and when I then click for an integral, the only error message is in the mantid logger. Will a user notice this and is the message helpful?