mantidproject/mantid #32277: Manual Testing Mantid Basics Course

martyngigg, 2021-08-20:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

* Build a docs-html target of Mantid
* Build docs-qtassistant and docs-qthelp so that autogenerated plots are available
* Check against the online documentation (http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/index.html#mantid-basic-course)
* Open up the basic course (docs/html/tutorials/mantid_basic_course/index.html)
* Check that the pages in there make sense

Harrietbrown, 2021-08-24:
On the loading data it lists `LOQ74019-74020` as an example of summing a range of runs into one workspace. however trying that raises an error

```Load started
Unable to successfully run LoadLog Child Algorithm: 
File \\isis.cclrc.ac.uk\inst$\ndxloq\instrument\data\cycle_99_4\LOQ74019.log cannot be read because it has an old unsupported format.
Cannot process ICPevent log. Period 1 assumed for all data.
Unable to successfully run LoadLog Child Algorithm: 
File \\isis.cclrc.ac.uk\inst$\ndxloq\instrument\data\cycle_99_4\LOQ74020.log cannot be read because it has an old unsupported format.
Cannot process ICPevent log. Period 1 assumed for all data.
Error in execution of algorithm Plus:
Left and right sides should contain the same amount of spectra or the right side should contain only one spectra.
Error in execution of algorithm Load:
Left and right sides should contain the same amount of spectra or the right side should contain only one spectra.```

DanielMurphy22, 2021-08-24:
I think this data is in the TrainingCourseData, (which is not explicitly mentioned for these examples) and when rewriting these annoying load examples I set ManageUserDirectories to look there, rather than on the archive, so that's why I hadn't seen this problem. This ICPevent stuff look familar and has been half-solved in the past. I'll see if I can find the last issue we tackled related to this.

martyngigg, 2021-08-24:
So I think the warnings about the log data are due to the changes we made in #30964. There is a point in the past where .log files for instruments were all different and we basically don't know how to read them. We confirmed with those who have looked at data like this that they don't want the logs so that part is not a problem.

The errors in `Plus` are a different story though.  Does 6.1 do the same and fail on the Plus?

Harrietbrown, 2021-08-24:
Loading using `LOQ74019-74020` works fine on 6.1

DanielMurphy22, 2021-08-24:
> Loading using `LOQ74019-74020` works fine on 6.1

I'm seeing `LOQ74019-74020` fail on 6.1 and 5.1.1

Harrietbrown, 2021-08-24:
> > Loading using `LOQ74019-74020` works fine on 6.1
> 
> I'm seeing `LOQ74019-74020` fail on 6.1 and 5.1.1 on Windows

Odd just tried again and got the erro. maybe I was running the wrong test by mistake.

DanielMurphy22, 2021-08-24:
Ah yes! In the last line of the [webpage](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/01_loading_data.html#loading-lots-of-data-files), I added the comment `For the last example, you may see an error, which correctly mentions that only data files with the same structure (number of spectra), should be summed.` I had added this as I couldn't find any datasets in our TrainingCourseData that met the requirement of having the same number of spectra AND had adjacent run numbers (74014 and 74020 would work if there wasn't a gap between their run numbers). I add this example despite the error as I thought it was more useful than the previous `INST1:INST2`.

Harrietbrown, 2021-08-25:
> Ah yes! In the last line of the [webpage](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/01_loading_data.html#loading-lots-of-data-files), I added the comment `For the last example, you may see an error, which correctly mentions that only data files with the same structure (number of spectra), should be summed.` I had added this as I couldn't find any datasets in our TrainingCourseData that met the requirement of having the same number of spectra AND had adjacent run numbers (74014 and 74020 would work if there wasn't a gap between their run numbers). I add this example despite the error as I thought it was more useful than the previous `INST1:INST2`.
> 
> Happy for another suggestion or if you think the broken example should be run numbers that aren't in the TrainingCourseData so that users don't run into this error?

I think It has to be data available to the users. if there is none available then we should give an example of the formatting but make it clear that the error they will get when testing it is expected.

DanielMurphy22, 2021-08-25:
[image]

If the user sets the SpectrumMax to 8 then the error is not encountered! I'll add a much better comment and make it more clear that it's related to this example :

`LOQ74019-74020` *

_*_ As these files have different numbers of spectra, you will see the error `Left and right sides should contain the same amount of spectra...`. For this example please load `LOQ74019-74020` with the parameter `SpectrumMax = 8`.

closed 2021-09-01 by DanielMurphy22