mantidproject/mantid #32278: Manual Testing M-slice

martyngigg, 2021-08-20:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See https://mantidproject.github.io/mslice/ and follow Quick Start, this should probably be tested in iDAaaS

robertapplin, 2021-08-24:
Testing on windows

robertapplin, 2021-08-24:
Issues found while testing:
#32296

robertapplin, 2021-08-24:
Closed because no more issues found

closed 2021-08-24 by robertapplin