mantidproject/mantid #32281: Manual Testing Sliceviewer

martyngigg, 2021-08-20:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

Test for both MDEvent and MDHisto workspaces: Briefly I tested for both MDEvent and MDHisto workspaces:
[ ] 1. Viewing 2D,3D and 4D MD data - (changing bins and sliders, color limits, colormap etc.)
[ ] 2. Panning, home, Zooming/Dynamic rebinning - both by selecting region with mouse and scrolling
[ ] 3. Toggling tracking of cursor
[ ] 4. Toggle gridlines
[ ] 5. Experiment with line plots and box integration (disabled fro non-orthogonal data)
[ ] 6. Changing viewing axes
[ ] 7. Peak overlay (multiple peaks workspaces, integrated peaks, sorting table, zooming on peaks, changing axes with peaks shown, peak overlay should not be shown for 2D data)
[ ] 8. Repeat (1-7) for data with non-orthogonal axes
[ ] 9. Play around with window - resizing, adjusting peak table size
[ ] 10. Saving plot
[ ] 11. Delete/rename underlying workspaces (at present sliceviewer does not update)

martyngigg, 2021-08-23:
It might be wise to wait until #32204 is merged before beginning manual testing of this.

martyngigg, 2021-08-23:
#32204 is held up as we think there could be a deeper issue. On balance I think it's worth starting testing out the sliceviewer without this fix.

DanielMurphy22, 2021-08-25:
The colorbar scale seems to have Log as an disabled option. Why and when is it enabled?

DanielMurphy22, 2021-08-25:
In [this section](https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html#line-subplots-and-region-of-interest-integration), trying to take a y cut by pressing the y key when the CNCS_7860_event workspace is opened in sliceviewer gives this error:

```
Error in execution of algorithm ExtractSpectra:
Invalid data found during construction of HistogramX
ExtractSpectra-v1: Invalid data found during construction of HistogramX
```

DanielMurphy22, 2021-08-25:
Small thing I noticed, for SXD_MD_nonortho, that in the non-orthogonal projection select the trifold toolbar button, if I edit the bin numbers then there is some white background that "appears" maybe showing the edge of the captured data. Just a little odd that this is not the case when the non-ortho projection is first displayed
[image]
[image]

DanielMurphy22, 2021-08-25:
A mediumly worrying issue, no crash or error, but I can get data to display in the wrong place (it moves but the axes values do not). OPen slicevier on SXD_MD_nonortho. Put K on the X axis and H on the Y, and also press the home toolbar button to reset the axes. Click the trifold Nonortho projection toolbar button. (see first image below is at this stage). Finally either adjust the L axis slider or it's thickness and notice the data has shifted to the left and zoomed little (but the axis values remain the same). To play around with this alternate clicking the home button and adjusting the thickness.

[image]
[image]

DanielMurphy22, 2021-08-25:
In Non-orthogonal view, the cursor value for signal is missing (just displays "-")

[image]

DanielMurphy22, 2021-08-25:
Should have mentioned I'm testing on a build of release-next + fixes for pythonlock up to I believe this commit: 80a6c15b0db6de61e6553a2f419c09a55e59c530

DanielMurphy22, 2021-08-25:
Peak overlay seems to generally work okay although given Remove peaks seems to remove the nearest peak and acts on the selected peaks_workspace (not neccessarily the peak you've clicked to view), it's possible for a user to unknowingly delete peaks from one peaks_ws while viewing and trying to remove the peak from another. But this is only possible when overlaying multiple peaks workspaces.

DanielMurphy22, 2021-08-25:
Using test data `mdHisto_3D` selecting a region of interest and trying to click the letter keys x,y,c,r to output slices all fail with error: `tuple indices must be integers or slices, not float`

DanielMurphy22, 2021-08-25:
I wonder why section 9 has the double asterisks at the end? https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html#resizing

DanielMurphy22, 2021-08-25:
Should add to section 5 of the Manual testing instructions about the Add/Remove peaks functionality

martyngigg, 2021-08-25:
> I wonder why section 9 has the double asterisks at the end? https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html#resizing

I think that's a misplaced end of bold region (`**`) that can be removed.

DanielMurphy22, 2021-08-25:
I know the testing instructions say that overlaying peaks is only for 3D workspace, but the peak overlay button is enabled for the `md_4D` workspace and in overlaying (really the wrong peaks anyway) you may get an error like this:
```
Traceback (most recent call last):
  File "/opt/mantidunstable/lib/mantidqt/widgets/sliceviewer/presenter.py", line 209, in slicepoint_changed
    PeaksViewerPresenter.Event.SlicePointChanged)
  File "/opt/mantidunstable/lib/mantidqt/widgets/sliceviewer/presenter.py", line 480, in _call_peaks_presenter_if_created
    getattr(self._peaks_presenter, attr)(*args, **kwargs)
  File "/opt/mantidunstable/lib/mantidqt/widgets/sliceviewer/peaksviewer/presenter.py", line 285, in notify
    presenter.notify(event)
  File "/opt/mantidunstable/lib/mantidqt/widgets/sliceviewer/peaksviewer/presenter.py", line 93, in notify
    self._overlay_peaks()
  File "/opt/mantidunstable/lib/mantidqt/widgets/sliceviewer/peaksviewer/presenter.py", line 123, in _overlay_peaks
    self.model.draw_peaks(self._view.sliceinfo, self._view.painter)
  File "/opt/mantidunstable/lib/mantidqt/widgets/sliceviewer/peaksviewer/model.py", line 85, in draw_peaks
    painter, self.fg_color, self.bg_color)
  File "/opt/mantidunstable/lib/mantidqt/widgets/sliceviewer/peaksviewer/representation/draw.py", line 61, in draw_peak_representation
    bg_color)
  File "/opt/mantidunstable/lib/mantidqt/widgets/sliceviewer/peaksviewer/representation/noshape.py", line 40, in draw
    peak_origin = slice_info.transform(peak_origin)
  File "/opt/mantidunstable/lib/mantidqt/widgets/sliceviewer/sliceinfo.py", line 89, in transform
    return np.array((*self._axes_tr(point[self._display_x], point[self._display_y]),
IndexError: index out of range
```

Furthermore if you change the axes to say have L on the X axis for the same `md_4D` workspace then the non-orthogonal projection button has incorrectly been enabled:
```
Traceback (most recent call last):
  File "/opt/mantidunstable/lib/mantidqt/widgets/sliceviewer/view.py", line 429, in on_non_orthogonal_axes_toggle
    self.presenter.nonorthogonal_axes(state)
  File "/opt/mantidunstable/lib/mantidqt/widgets/sliceviewer/presenter.py", line 354, in nonorthogonal_axes
    self.model.create_nonorthogonal_transform(self.get_sliceinfo()))
  File "/opt/mantidunstable/lib/mantidqt/widgets/sliceviewer/model.py", line 262, in create_nonorthogonal_transform
    raise RuntimeError("Workspace cannot support non-orthogonal axes view")
RuntimeError: Workspace cannot support non-orthogonal axes view
```

martyngigg, 2021-08-25:
> I know the testing instructions say that overlaying peaks is only for 3D workspace, but the peak overlay button is enabled for the `md_4D` workspace and in overlaying (really the wrong peaks anyway) [...]

I couldn't reproduce this so could depend a lot on the mix of data/peaks that are used.

> Furthermore if you change the axes to say have L on the X axis for the same `md_4D` workspace then the non-orthogonal projection button has incorrectly been enabled:
>
> ```
> Traceback (most recent call [...]

This is a bug. The state of the button must not be altered based on the axis change. Can you raise an issue against core for 6.3S1? I don't think it's high enough priority to fix now.

DanielMurphy22, 2021-08-25:
> This is a bug. The state of the button must not be altered based on the axis change. Can you raise an issue against core for 6.3S1? I don't think it's high enough priority to fix now.

This existed in release 6.1 so appropriate for 6.3 S1

closed 2021-08-25 by DanielMurphy22