mantidproject/mantid #32306: Project Recovery crash when closing mantid on windows

robertapplin, 2021-08-25:
**Describe the bug**
There is a crash which sometimes happens after clicking the `X` to close mantid. It is some kind of issue to do with project recovery. It might be a windows-only issue.

I haven't been able to reproduce it reliably, but I think it warrants investigation.

**To Reproduce**
The general sequence of events was:
1. Open Mantid and set the Project recovery time between recovery checkpoints to 2 seconds in `Settings`
2. Run a script to create a workspace
```py
CreateWorkspace(DataX=range(12), DataY=range(12), DataE=range(12), NSpec=4, OutputWorkspace='NewWorkspace')
```
3. Cause Mantid to crash using the `Segfault` algorithm
4. Open Mantid again. I am not sure if I started Mantid normally or performed a recovery
5. Close Mantid using the `X` and the sad mantid appears. This error was seen in the logger
[image]

I didn't manage to reproduce this error, but it looks like the problem occurs when trying to delete a directory that isn't empty.

**Platform/Version (please complete the following information):**
 - OS: Windows 10
 - Mantid Version 6.1.2

robertapplin, 2021-08-25:
I am setting the milestone to 6.2, but this is to be decided by the core team. I realise there's not much info to go by especially as I couldn't reproduce it, but its probably worth investigating.

StephenSmith25, 2021-08-25:
I feel like this is something that has cropped up numerous times #32065 #28184 #25469 (possibly are all related). I guess the struggle is reliably reproducing it.

robertapplin, 2021-08-25:
Yes it seems like quite a common issue but hard to reproduce. Closing because this is a duplicate of #32065 . I will comment about this issue in #32065 .

closed 2021-08-25 by robertapplin