mantidproject/mantid #32317: SliceViewer data shifts when thickness is adjusted in non-ortho projection

DanielMurphy22, 2021-08-25:
Originally reported in Manual testing : https://github.com/mantidproject/mantid/issues/32281#issuecomment-905462572

**Describe the bug**
On Sliceviewer for Nonorthogonal data, I can get data to display in the wrong place (it moves but the axes values do not).

**To Reproduce**
- Run this script:
```
from mantid.simpleapi import *
# SXD23767.raw is available in the TrainingCourseData from the downloads page
SXD23767 = Load(Filename='SXD23767.raw', LoadMonitors='Exclude')
# Set some UB with angles we can play with
SetUB(SXD23767, 1,1,2,90,90,120)
SXD_MD_nonortho = ConvertToDiffractionMDWorkspace(InputWorkspace='SXD23767', OutputDimensions='HKL')
```

- Open sliceviewer on the workspace `SXD_MD_nonortho`. 
- Put K on the X axis and H on the Y, and also press the home toolbar button to reset the axes. 
- Click the trifold Nonortho projection toolbar button. 
[image]

- Finally either adjust the L axis slider or it's thickness and notice the data has shifted to the left and zoomed a little (but the axis values remain the same). To play around with this alternate clicking the home button and adjusting the thickness.
[image]

**Expected behavior**
The data does not shift when the L thickness or L slider is adjusted

**Platform/Version (please complete the following information):**
 - OS: IDAaaS (RHEL7) and Windows 10
 - Mantid Version pre-beta release of 6.2, also exists in 6.1 and 5.1.1

**Additional context**
In the pre-beta release of 6.2, this data moving problem exists only in the non-orthogonal view, but in releases 6.1 and 5.1.1, the problem existed in both non-ortho and normal view, so in the last dev cycle we’ve fixed this for normal view somehow.

RichardWaiteSTFC, 2021-09-02:
Interestingly I can reproduce this with SXD data but not with fake data (Event or histo) made a so
```
ws = CreateMDWorkspace(Dimensions='3', Extents='-6,6,-4,4,-0.5,0.5',
                       Names='H,K,L', Units='r.l.u.,r.l.u.,r.l.u.',
                       Frames='HKL,HKL,HKL',
                       SplitInto='2', SplitThreshold='50')
print(ws.getSpecialCoordinateSystem())
expt_info = CreateSampleWorkspace()
ws.addExperimentInfo(expt_info)
SetUB(ws, 1,1,2,90,90,120)
FakeMDEventData(ws, PeakParams='1e+06,0,0,0,1', RandomSeed='3873875')
 
# try with MDHistoToWorkspace
BinMD(InputWorkspace='ws', AxisAligned=False, 
    BasisVector0='H,r.l.u.,1.0,0.0,0.0', 
    BasisVector1='K,r.l.u.,0.0,1.0,0.0', 
    BasisVector2='L,r.l.u.,0.0,0.0,1.0',
    OutputExtents='-6,6,-4,4,-0.5,0.5', OutputBins='100,100,100', 
    NormalizeBasisVectors=False, OutputWorkspace='ws_histo')
```

martyngigg, 2021-09-03:
This is important to fix so we're including it in 6.2.

RichardWaiteSTFC, 2021-09-03:
> This is important to fix so we're including it in 6.2.

@martyngigg can we also move #32314 to 6.2, it can cause an exception and I've already found a fix

martyngigg, 2021-09-03:
> @martyngigg can we also move #32314 to 6.2, it can cause an exception and I've already found a fix

That's fine. Please create the pull request against `release-next` and then we can assess to risk and move if we feel uncomfortable.

martyngigg, 2021-09-03:
Fixed by #32375

PR #32375 (2021-09-03): Update nonorthog plot with correct limits in sliceviewer
**Description of work.**

Fixed bug whereby plot was getting updated with the orthog limits in nonorthog mode (but not in new plot hence clicking the home button corrected the plot).

**To test:**

(1) Follow instructions in original issue
(2) Check limits and zooming etc. work on orthogonal workspaces (and non-orthog in orthog mode) and matrix workspaces

Can create event and histo workspaces like so
```
ws = CreateMDWorkspace(Dimensions='3', Extents='-6,6,-4,4,-0.5,0.5',
                       Names='H,K,L', Units='r.l.u.,r.l.u.,r.l.u. [... cut]
closed 2021-09-03 by martyngigg