mantidproject/mantid #32318: SliceViewer Nonortho Cursor Image Info Signal is -

DanielMurphy22, 2021-08-25:
**Original reporter:** Manual Testing: https://github.com/mantidproject/mantid/issues/32281#issuecomment-905466044

**Describe the bug**
There is no signal in the cursor Image info widget in the non-orthogonal view. This signal is just displayed as `-`.

**To Reproduce**
 - Put the TrainingCourseData in ManageUSerDirectories (or be connected to the ISIS data archive).
 - Run this script:
```
from mantid.simpleapi import *

# SXD23767.raw is available in the TrainingCourseData from the downloads page
SXD23767 = Load(Filename='SXD23767.raw', LoadMonitors='Exclude')
# Set some UB with angles we can play with
SetUB(SXD23767, 1,1,2,90,90,120)
SXD_MD_nonortho = ConvertToDiffractionMDWorkspace(InputWorkspace='SXD23767', OutputDimensions='HKL')
```
- Right-click on the `SXD_MD_nonortho` workspace and click Show Sliceviewer
- Click the trifold non-orthogonal projection toolbar button and hover the cursor over the image
- Notice that the Signal value in the cursor image info widget is `-`.

**Expected behavior**
The signal is displayed in the widget as it is for orthogonal view ( or if we don't want to then remove the signal column for MD workspaces).

**Screenshots**
[image]

**Platform/Version (please complete the following information):**
 - OS: Rhel7 (IDAaaS) and Windows 10
 - Mantid Version pre-beta 6.2, and releases 6.1 and 5.1.1

RichardWaiteSTFC, 2021-09-01:
I think this might be deliberate, because it's set at DBL_MAX when non-orthog view is enabled it calls `_on_cursor_at_mesh` rather than `_on_cursor_at_axesimage`.
https://github.com/mantidproject/mantid/blob/77ea7fe3ace7c873f672a2f2663aef3eb5eab75e/qt/python/mantidqt/widgets/sliceviewer/imageinfowidget.py#L76-L89

sf1919, 2023-02-08:
If this is deliberate should this one be closed? @thomashampson @RichardWaiteSTFC

thomashampson, 2023-02-10:
@DanielMurphy22 had the suggestion of removing the signal column if it's not applicable for this case.

PR #36926 (2024-04-11): Hide signal value in sliceview non-orthog view
### Description of work

Previously in the Image Info Widget, Signal has been `-` in non-orthogonal view. This Pr hides the signal column, as suggested in the attached issue.

Fixes #32318

### To test:

- Run the script from the issue and open in the slice viewer
- Toggling non-orthogonal view should hide / show the signal column in the image info widget.
- Try some other tests, like closing the sliceviewer with non-orthogonal view enabled and reopening with different data.

---

### Reviewer

Please comment on the points listed below ([full desc [... cut]
closed 2024-04-11 by robertapplin