mantidproject/mantid #32359: SliceViewer z-axis autoscale lag

thomashampson, 2021-09-01:
**Describe the bug**
Autoscaling of the z(colour)-axis lags behind zoom operations by one. However, it seems to work fine for event workspaces.

**To Reproduce**
Steps to reproduce the behavior:
1. Load workspace and open the slice viewer.
2. Make sure "Autoscaling" is ticked (bottom right of the SliceViewer window).
3. Zoom into a region of low occupancy using the "zoom to rectangle" tool.
4. The limits of the z-axis will remain the same until you perform another zoom operation, e.g. scroll once into the plot.
5. Click the home button to reset the x- and y-axis limts.
6. The limits of the z-axis will not update to account for the range of the data until you click the home button **again**.

**Expected behavior**
z-axis should recale correctly after each zoom operation.

**Platform/Version (please complete the following information):**
 - OS: All
 - Mantid Version 6.1

**Additional context**

gemmaguest, 2022-01-12:
See comment on #33214.

Note that I see exactly the same problem with both matrix and event workspaces.

closed 2022-07-21 by stale[bot]

thomashampson, 2022-11-01:
This has been fixed