mantidproject/mantid #32369: Investigate moving interfaces out of the scripts folder

StephenSmith25, 2021-09-02:
## Describe the outcome that is desired.

Move the scientific interfaces out of the scripts folder into a folder within the qt directory.

## Does the outcome relate directly to a problem? Please describe.

It doesn't make sense to have interfaces in the scripts folder. The scripts folder is part of a framework build, which doesn't include mantidqt. The interfaces are dependent on mantidqt, so shouldn't be bundled into a framework build or install of mantid. 

## Describe any solutions you are considering

Move the interfaces into a python package defined in `qt/scientific_interfaces/python/NAME_OF_PACKAGE`

```
├── qt/scientific_interfaces/python
│   ├── scientific_interfaces (name of the python package)
│          └── __init__.py
│          └── Muon
│          └── ..
│   └── setup.py
```
Contents of setup.py
```
setup(
    name='scientific_interfaces',  # This is likely not a good name for a python package unless we are restricted to internal usage. 
    install_requires=['mantidqt'],
    version=os.environ['MANTID_VERSION_STR'],
    packages=find_packages(exclude=['*.test']),
    package_data={'': ['*.ui']},
)
```

Imports from a scientific interface are changed to
`from Muon.GUI.` goes to  `from scientific_interfaces.Muon.GUI`

We'll have to add `qt/scientific_interfaces/python` to the python path, e.g. through one of the .pth files.

## Questions:

1) Does the structure make sense?
2) Is scientific_interfaces a good package name? For internal use it makes sense, but if somebody conda installed or pip installed mantid a package called scientific_interfaces feels a bit ambiguous.
3) Where do we install the interfaces as they are no longer in the installed scripts folder. Do they go in `plugins/python`?

StephenSmith25, 2021-09-02:
Could I get some opinions from you @gvardany @martyngigg @DavidFair @peterfpeterson so we can get this moving during the current maintenance window.

martyngigg, 2021-09-02:
I'd be happy with `qt/python/scientific_interfaces` or we could go with a Stephen's suggestion`qt/scientific_interfaces/python` and look to a slight restructure of the qt directory anyway as it's a little confusing. If we are accepting our public-facing api is Python then we could look to move the C++ bits to more internal locations and dispense with these python directories that I know people find confusing.

Something flatter like:
```
qt/
  -- common/ (C++ libraries that are used by lots of other things: CommonWidgets, MplCpp, InstrumentView) 
  -- mantidqt/
  -- scientific_interfaces/
      -- isis_reflectometry (C++ with Python wrapper to launch)
      -- muon (pure python)
      -- ...
  -- workbench/
```
?

martyngigg, 2021-09-02:
On the Python package name. Do we need to provide a conda package for it (yet?) If we can't make a decision on it then maybe we keep it "private" for the moment?

StephenSmith25, 2021-09-02:
I agree that `qt/python/scientific_interfaces` is a good place. In terms of the package name, I think it does make sense to choose a good name now, even if we aren't shipping anything, as the package name is/should be the first thing we type when we are importing something in python. At the moment, when we are importing interfaces in the script folder we make imports like

`from Muon.GUI... import ..`
which only works because we've added the root of this package to some .pth file. I don't feel its very obvious where `Muon.GUI` is defined and you definitely wouldn't start your search in the scripts folder. Comparing this to importing the icons, 

`from mantidqt import icons`,

its very obvious that the icons are part of the mantidqt python package. I guess my issue is the interfaces are currently a bunch of directories in the scripts folder, rather than a python package which would have a consistent import prefix, e.g
```
from my_interfaces_pacakge.Muon.GUI import *
from my_interfaces_package.Engineering_diffraction import *
```
I think this would be clearer for new developers where they are sourcing their packages/functionality from.

martyngigg, 2021-09-02:
I definitely don't disagree. I was just concerned at holding up the work over it as it can be changed fairly easily if we are not deploying a public package.

In line with everything else we should probably have `mantid` in it? When I think about it I can't seem to get far away from the words science/scientific and interfaces. A brain dump (I'm basically just writing them as they fall out of my head)
 
- mantid_science_interfaces
- mantid_technique_interfaces
- mantid_workflow_guis
- mantid_science_workflow_interfaces
- mantid_reduction_interfaces (although Muon/ISIS Inelastic due analysis too)
- mantid_reduction_guis
- some variation on this?

StephenSmith25, 2021-09-02:
Yeah I agree, mantid should be in the name of the package, but would mantid_interfaces suffice? Just trying to make it as short as possible to reduce the typo rate. 

Then we have three mantid package imports
```
from mantid import api
from mantidqt import icons
from mantid_interfaces import Muon
```
If we are placing it in qt/python we may need to restructure the folder as there is already a `setup.py` file in that folder used to define the mantidqt package. I think something like:

```
├── qt/python
│   ├── mantidqt (this folder contains our mantidqt python pacakge)
│          └── setup.py # setup file for our mantidqt project
│          └── mantidqt # folder containing mantdqt sources
|                  └── __init__.py 
|                  └── ....
│   ├── mantid_interfaces (this folder contains our mantid interfaces python package)
│          └── setup.py # setup file for our mantid_interfaces project
│          └── mantid_interfaces  # folder containing mantid_interfaces sources
│                  └── __init__.py 
|                  └── ....
```
keeps a standard layout for each package, meaning we can get `setup.py` to do most the work for us.

gvardany, 2021-09-02:
Agreed with the structure. I was about to suggest a longer name `mantid_plugin_interfaces`, if we want to keep shorter, `mantidguis` is probably the shortest. Whatever the name we end up, it should probably match with the C++ counterpart `qt/scientific_interfaces`, to be consistent with widgets.

martyngigg, 2021-09-06:
How about `mantidqtinterfaces`? It builds on the existing package with that name and is still reasonably short...

StephenSmith25, 2021-09-06:
Yeah, mantidqtinterfaces works well - I'll go with that

peterfpeterson, 2021-09-07:
Why aren't these in subpackages, e.g. `mantid.qt` and `mantid.qt.interfaces` or `mantid.qt` and `mantid.qtinterfaces`?

StephenSmith25, 2021-09-07:
I had thought about this. We could define mantid as a namespace package, i.e. following the python packaging specifications : https://packaging.python.org/guides/packaging-namespace-packages/. But I was slightly apprehensive as its quite a big redesign (probably more than necessary at the moment.)

martyngigg, 2021-09-07:
I remember us discussing this a few years ago and we were uncertain at how some package managers like rpm would handle sharing directory ownership so we went along with separate packages. I think in my mind I had also thought, especially with the interfaces, that keeping them built "on top" might be preferable if we decided to move them to a separate repo say but this feels like a weak argument as I type it.

As Stephen says, I am concerned this would be quite a big redesign and would the subpackages over substantive benefits?

peterfpeterson, 2021-09-07:
The `mantidqtinterfaces` plan works for me too. I was asking now because it is easier to change only once.