mantidproject/mantid #32377: Crash when changing Muon Instrument after PhaseQuad

sf1919, 2021-09-03:
**Original reporter:** Peter Baker ISIS

**Describe the bug**
After generating a PhaseQuad Table for a run if you then try to change the Instrument to load a different run and error occurs that launches the Error Reporter. This happens in both the Muon Analysis and the Frequency Domain Analysis. This issue was observed on Windows using workbench 6.1.20210831.1435. It was also recreated using 6.1.20210902.1514 which means #32344 does not resolve this.

The stack trace for the crash is as follows

```Stack Trace:
       Traceback (most recent call last):
 File "....\scripts\Muon\GUI\Common\phase_table_widget\phase_table_presenter.py", line 239, in handle_phase_table_changed
   self.context.calculate_all_pairs()  # Updates phasequads
AttributeError: 'FrequencyDomainAnalysisContext' object has no attribute 'calculate_all_pairs'```

**To Reproduce**
On Muon Analysis
1. Open Muon Analysis GUI
2. Set instrument to MUSR on the home tab and load run 62260
3. Click on the Phase Table Tab and click Calculate Phase Table. Don't bother giving the table a name (uses default instead).
4. Once Phase Table has been generated return to the Home Tab and change the instrument. Error occurs at this point.

On Frequency Domain Analysis
1. Open Frequency Domain Analysis GUI
2. Set instrument to EMU on the home tab and load run 85022
3. Click on the Phase Table Tab and click Calculate Phase Table. Don't bother giving the table a name (uses default instead).
4. Once Phase Table has been generated return to the Home Tab and change the instrument. Error occurs at this point.

**Expected behavior**
When changing intstrument to load a different run after working on a previous run that has a PhaseQuad Table there should be no issues.

**Platform/Version (please complete the following information):**
 - OS: Windows (but likely to occur on other platforms
 - OS Version: 10
 - Mantid Version: 6.1.20210831.1435

AnthonyLim23, 2021-09-06:
Just need to connect the phasequad table to the change instrument notifier so that it gets cleared. The bug is that it is trying to create a phasequad with no data.

martyngigg, 2021-09-08:
Fixed by #32415

closed 2021-09-08 by martyngigg