mantidproject/mantid #32388: Muon model analysis fit bugs

AnthonyLim23, 2021-09-06:
Original reporter: James

Describe the bug

In muon analysis there are a couple of bugs in the model analysis tab.

The first is that it ignores the data that has values copied to it via a tie/global. This is due to the IgnoreInvalidData being set to True (for BG corrections)

For MUSR62260-5, in the model analysis. Setting the x axis to slits makes the fitting range nonsense. As the precision of the range is lower than that of the data.

To Reproduce

Open muon analysis
Load MUSR 62260
Go to fitting and click simultaneous
Add a gausosc (set everything except Phi to global)
Set the frequency to 1
Click Fit

Step to the next run (62261)
Step the parameters to the next group (using the data selector in the fitting tab)
Click fit

Step to the next run (62262)
Step the parameters to the next group (using the data selector in the fitting tab)
Click fit

Step to the next run (62263)
Step the parameters to the next group (using the data selector in the fitting tab)
Click fit

Go to the results table tab
In the logs table tick "Slits"
Click create table

Go to the model analysis tab
Set the x axis to Slits and y to A
Add a fit function
Press fit
The start and end X are identical and they shouldn't be.
The fit will ignore 3 of the 4 data points.

Expected behavior

Screenshots

Platform/Version (please complete the following information):

OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
OS Version:
Mantid Version [e.g. 6.0.0]
Additional context

AnthonyLim23, 2021-09-13:
@sf1919 can you check which of the above bugs still exist?

sf1919, 2021-09-14:
I think the first of the issues has been resolved as four things were plotted.

As for the second issue - the x min and x max are very close values to one another. Not sure what was originally meant by 'makes the fitting range nonsense'. Below is the screenshot of what I get when I following the steps above.

[image]

robertapplin, 2021-09-14:
The second issue might have been fixed by #32443

AnthonyLim23, 2021-09-14:
What I meant was that start X = end X, which is nonsense

sf1919, 2021-09-14:
Well the Start X and End X are not identical.... close but not identical.... theres 0.003 in it.

Looks like we can probably close this one afterall.

PR #32440 (2021-09-09): Result table global parameters should have a non-zero error
**Description of work.**
This PR fixes a bug where the global parameter saved in a results table can sometimes have a zero error arbitrarily. This does not make complete sense as it technically does have an error as it is tied to another parameter which has an error.

This PR makes sure the global parameter with the non-zero error value is used. As a result, a bug found by James is also fixed:
```
I set up a simultaneous fit (e.g. MUSR run 62260, choose Simultaneous across run, choose 
function GausOsc, set A, Frequency, Sigma to Glob [... cut]
closed 2021-09-24 by AnthonyLim23