mantidproject/mantid #32414: Indirect Data Anaysis fit tab table values unreadable

thomashampson, 2021-09-08:
**Describe the bug**
The StartX and EndXFit table values are unreadable in the Fit tabs of the Inelastic Data Analyis interface, except when the column widths are increased to fit the entire float. The value is only visibile if the entire float fits inside the box, otherwise it will say "0...." or "1....".

**To Reproduce**
1. Go to Interfaces > Indirect > Data Analysis
2. Go to the "I(Q, t) Fit" tab
3. Click "Add Workspace" and load some data (e.g. mantid-build/ExternalData/Testing/Data/UnitTest/iris26176_graphite002_iqt.nxs)
4. Enter a workspace index or tick "all spectra" and click "Add" then "Close"
5. Drag the range sliders around on the mini plot and examine the StartX and EndX values in the table

**Expected behavior**
Boxes should display as many characters as possible before the "..."

**Screenshots**
Current behaviour:
[image]
Correct behaviour:
[image]

**Platform/Version (please complete the following information):**
 - OS: Windows 10
 - Mantid Version: occurs on latest build of release next (6.1.20210907.1448) but not on a ~1 week old nightly (6.1.20210831.1435).

SpencerHowells, 2021-09-08:
@thomashampson @Harrietbrown is dealing with a bug in the StartX/EndX values in PR #32394

Harrietbrown, 2021-09-09:
I am not getting this issue in my build, 
[image]
I am not sure what is happenign for you

thomashampson, 2021-09-13:
This seems to only occur on my locally built version of release-next, and not on the deployed nightly version. Therefore I'll mark it as closed because it won't affect any users.

closed 2021-09-13 by thomashampson