mantidproject/mantid #32416: Indirect Data Analysis crashes Mantid if wrong file is loaded

thomashampson, 2021-09-08:
**Describe the bug**
In the Elwin tab of the Indirect Data Analysis interface, if you load data that it isn't expecting, it will cause Mantid to crash.

**To Reproduce**
1. Open the interface: Interfaces > Indirect > Data Analysis
2. Staying on the Elwin tab, click "Browse" to open the file dialog.
3. Change the file extension filter to "All files (*)"
4. Load, for example, EMU00020884.nxs from the training course data
5. Mantid will crash without any stack trace

**Expected behavior**
Trying to load data that isn't expected should be handled more gracefully.

**Platform/Version (please complete the following information):**
 - OS: Windows 10
 - Mantid Version 6.1 onwards (and probably earlier)

SpencerHowells, 2021-09-08:
@thomashampson EMU is not an inelastic neutron instrument so it’s not surprising that it crashes! Check the IDF so that it says inelastic & if not give error message that it’s invalid instrument.

SpencerHowells, 2021-09-08:
@thomashampson In the Browse dialog it searches for files with the suffix _iqt to prevent users trying to read stupid data. If the user chooses to ignore this then it’s their problem.

SpencerHowells, 2021-09-08:
@sf1919 If this needs to be done then it has to be in every tab in every interface!

Harrietbrown, 2021-09-08:
> @sf1919 If this needs to be done then it has to be in every tab in every interface!

Only for cases where it causes a hard crash without an error report screen.

SpencerHowells, 2021-09-08:
@Harrietbrown This occurred when reading an EMU muon file so it is rubbish for our interfaces - all of them. You will need to check for this every where. This is getting stupid!! Who in their right mind would think of reading EMU for neutrons!

SpencerHowells, 2021-09-09:
@Harrietbrown @sf1919 Just had a look at it. If I load an IRIS raw file it comes back with an error message that says that efixed has not been defined. This is what you expect if it is incorrect data - either it is not inelastic neutron or efixed was not defined when creating the file. However if you try to delete the workspace Mantid crashes.

SpencerHowells, 2021-09-09:
@Harrietbrown @sf1919 Next look. I loaded a _red file & it plotted Ok. Again though I cannot remove it & it hangs. Eventually I managed to close the tab without crashing Mantid.

sf1919, 2021-09-16:
Closed by #32483

PR #32483 (2021-09-15): Prevented crash from incorrect file load in Elwin
**Description of work.**
The Elwin file load and the file loading in the add workspace dialog now check whether the file being loaded is of the correct type. If not loading is stopped and the user receives an error message. Mantid does not crash and the user can continue to use the Elwin interface.

**To test:**
Test file load
1. Open ```Indirect->Data Analysis``` GUI
2. Click Browse button
3. Find a file that does not have a ```_red``` or ```_sqw``` ending - for example ```EMU00020884.nxs``` from the Training Course data
4. You should receive  [... cut]
closed 2021-09-16 by sf1919