mantidproject/mantid #32437: Make Elwin Tab components dockable

sf1919, 2021-09-09:
The new load widget for the Elwin Tab takes up a lot of space causing the plotting area to be smaller than desired. 

**Describe the solution you'd like**
One solution that has been suggested is to make the components of the Elwin Tab dockable (like MSD Fit, I(Q, t) Fit, ConvFit, F(Q) Fit).

sf1919, 2022-03-08:
@Harrietbrown @AnthonyLim23 Do we still want to make the components dockable or has the issue of the load widget taking up a lot of space been resolved in #32475?

Harrietbrown, 2022-03-08:
Is the new workspace input tab able to completly replace the file input now? if so it would be best to remove that tab and that would help fix the space issue.

closed 2022-09-16 by stale[bot]