mantidproject/mantid #32439: Fitting bugs

AnthonyLim23, 2021-09-09:
**Describe the bug**

When attempting to do a fit from a plot in Mantid I have ran into a few problems.
1. I cannot get the fit to include the first data point
2. I cannot drag the fit range lines
3. If I update the endX in the fit browser, the line on the plot does not update (even after a fit)

**To Reproduce**

Load some data, then plot it (I just made a 10 data point ws)
Switch the view from line to points
Try to update the start/end x values
Try to include the first/last data point (you can't)

**Expected behavior**

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

StephenSmith25, 2021-09-09:
Maybe needs some more info. As this works for me, but there are some usability issues.

AnthonyLim23, 2021-09-09:
I ran the script below and when I tried to do the fitting I ran into the above problems.

```
PlotAsymmetryByLogValue(OutputWorkspace='data', LogValue='sample_magn_field', DeadTimeCorrType='FromRunData', WorkspaceNames=r'\\isis.cclrc.ac.uk\inst$\ndxhifi\instrument\data\cycle_14_3\HIFI00077776.nxs,\\isis.cclrc.ac.uk\inst$\ndxhifi\instrument\data\cycle_14_3\HIFI00077777.nxs,\\isis.cclrc.ac.uk\inst$\ndxhifi\instrument\data\cycle_14_3\HIFI00077778.nxs,\\isis.cclrc.ac.uk\inst$\ndxhifi\instrument\data\cycle_14_3\HIFI00077779.nxs,\\isis.cclrc.ac.uk\inst$\ndxhifi\instrument\data\cycle_14_3\HIFI00077780.nxs,\\isis.cclrc.ac.uk\inst$\ndxhifi\instrument\data\cycle_14_3\HIFI00077781.nxs,\\isis.cclrc.ac.uk\inst$\ndxhifi\instrument\data\cycle_14_3\HIFI00077782.nxs,\\isis.cclrc.ac.uk\inst$\ndxhifi\instrument\data\cycle_14_3\HIFI00077783.nxs,\\isis.cclrc.ac.uk\inst$\ndxhifi\instrument\data\cycle_14_3\HIFI00077784.nxs,\\isis.cclrc.ac.uk\inst$\ndxhifi\instrument\data\cycle_14_3\HIFI00077785.nxs,\\isis.cclrc.ac.uk\inst$\ndxhifi\instrument\data\cycle_14_3\HIFI00077786.nxs')

```

StephenSmith25, 2021-09-09:
The range issues I think are a usability thing, so maybe not as important. But for me the parameters aren't updating as well. Seems like we've got a issue here....

DavidFair, 2021-09-10:
I've removed high priority for this meta issue, as the params not updating has been resolved.
I'm not sure if the other issues are outstanding

martyngigg, 2021-09-10:
Fixed by #32457

closed 2021-09-10 by martyngigg

AnthonyLim23, 2021-09-13:
@StephenSmith25 does the PR also fix the problem of not being able to select the whole range? It also seems to be a problem in inherited fit browsers https://github.com/mantidproject/mantid/issues/32478

StephenSmith25, 2021-09-13:
So maybe I'm missing something, but I appear to be able to fit to all the data, e.g - or is this not what you mean - as I was able to do this before.

[image]

AnthonyLim23, 2021-09-13:
When I used points for the plot, I was unable to get the first or last. @Harrietbrown has seen the same problem in indirect

StephenSmith25, 2021-09-13:
Did you set startX and endX to the bin dimensions, rather than the coordinates of the points / bin centres?

AnthonyLim23, 2021-09-13:
I couldn't even drag the fitting line to it

StephenSmith25, 2021-09-13:
You can still set the startX and endX in the fit browser though. If you set the startX to the first bin and endX to the last bin I observe that it fits the whole range - so from what I can tell there's no bug here, but I maybe missing something. I agree there's possibly some issues with the fit range markers though.

AnthonyLim23, 2021-09-13:
I just tried that on the data from the script above. I changed the start x and it did the fit as expected. I changed the end x and it moved the start x to exclude the first point again

StephenSmith25, 2021-09-13:
Moving the endX doesn't move the startX for me 😕

AnthonyLim23, 2021-09-13:
it is inconsistent, it just worked for me. Something does not seem right, but its hard to pinpoint

StephenSmith25, 2021-09-13:
I think there's some usability bugs with the markers, which we should look at next release. But the startX and endX properties of the fit algorithm seem to work as intended, well as far as I can tell.

PR #32457 (2021-09-10): Fix fit parameters not updating in fit browser
**Description of work.**
This PR fixes an issue where the parameters from a fit were not updating in the fit browser. This was only an issue for single functions, as the parameters from the table were unprefixed, but due to design the fit browser, it treats every function as a composite function. So when it went to set the parameter values, using the table parameter names it threw (which was caught by the fit browser without any warnings or information). I believe a better way of updating the parameters is to use the output of the fit algorithm, r [... cut]