mantidproject/mantid #32442: Fitting in Muon GUIs sometimes excludes a data point due to GUI precision

robertapplin, 2021-09-09:
**Original reporter:** James Lorde

**Describe the bug**
From James:
```
In Model Fitting the default is to fit all the data and I can usually select the fit range properly. 
But with the above set of runs (62260-62265) and fitting versus Slits, if I set either limit it 
truncates the actual fit range in the plot at both ends, while the numbers are apparently 
truncated at the full data range. Other log values or parameters used as X don’t seem to 
trigger this bug. Maybe it’s a floating point rounding error?
```

The precision of the muon GUI can sometimes means  that some data points are cut-off and not included in a fit by accident. The solution is to offset the calculated x lower and x higher by a the precision used in muon GUIs.

**To Reproduce**
Open muon analysis
Load MUSR 62260
Go to fitting and click simultaneous
Add a gausosc (set everything except Phi to global)
Set the frequency to 1
Click Fit

Step to the next run (62261)
Step the parameters to the next group (using the data selector in the fitting tab)
Click fit

Step to the next run (62262)
Step the parameters to the next group (using the data selector in the fitting tab)
Click fit

Step to the next run (62263)
Step the parameters to the next group (using the data selector in the fitting tab)
Click fit

Go to the results table tab
In the logs table tick "Slits"
Click create table

Go to the model analysis tab
Set the x axis to Slits and y to A
Add a Linear Background
Press fit

You can see the first data point is being ignored by the fit.

[image]

**Platform/Version (please complete the following information):**
All

AnthonyLim23, 2021-09-09:
this bug is also part of #32388

DavidFair, 2021-09-10:
Fixed by #32443

PR #32443 (2021-09-10): Offset the x range of a fit in muon GUIs
**Description of work.**
This PR fixes an issue where the x range of a fit is sometimes not allowed to encompass the entire range of the actual data in the x axis. This is due to the GUI only allowing a certain precision, and so some data points at the start or end of the data can be cut off when doing a fit.

**To test:**
Follow the instructions in #32442 . Ensure that the first data point is included in the fit. The start X should be 4.599

Fixes #32442

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidpro [... cut]
closed 2021-09-10 by DavidFair