mantidproject/mantid #32454: Error in FDA when changing parameter value

robertapplin, 2021-09-09:
**Original reporter:** James Lord

**Describe the bug**
An error occurs after changing a parameter in the fitting tab of FDA when the sequential tab is being filtered.

**To Reproduce**
1. Frequency Domain Analysis interface
2. Load HIFI 77767-68
3. On Transform tab, click `Calculate FFT`
4. On Fitting tab, add a Gaussian function
4. On Sequential tab, change the `Filter By` option to show only Real data.
5. On fitting tab, change the `PeakCentre` parameter of the modulus data to a different value
6. You get this error
[image]

**Platform/Version (please complete the following information):**
All I assume

AnthonyLim23, 2021-09-10:
Looking at the instructions, this happens because the modulus is not in the filtered list. We should either:

a. Change the options in the selector on the fitting tab to match the filtered results
b. When the selector is changed to something that is not in the filter, turn the filter off

The filter was added in this PR:
https://github.com/mantidproject/mantid/pull/32173/files

martyngigg, 2021-09-10:
Fixed by #32468

PR #32468 (2021-09-10): Fix crash on Frequency Domain Analysis UI when updating fit function parameters
**Description of work.**

Fix a crash if the user updates the parameters on a fit function on the Fitting tab that isn't displayed on the Sequential tab due to a filter being in place (on the Sequential tab). The code that handles the fitting function update in scripts\Muon\GUI\Common\seq_fitting_tab_widget\seq_fitting_tab_presenter.py now only tries to update fit functions that are displayed in the table on Sequential tab

**To test:**

Follow instruction on linked issue

Fixes #32454.

*This does not require rele [... cut]
closed 2021-09-10 by martyngigg