mantidproject/mantid #32455: MuonBasePair object has no attribute forward_group error

robertapplin, 2021-09-09:
**Original reporter:** James Lord

**Describe the bug**
An error message `MuonBasePair object has no attribute forward_group error` appears when stepping to the next run after creating a phasequad.

**Reproduce this**
1. Open Frequency Domain Analysis interface
2. Load MUSR 62260
3. On Phase Table tab click `Calculate Phase Table` and then click the `+` at the bottom. Choose any name for them
4. On Transform tab, select the real and imaginary parts of the phase quad you just created.
5. Click `Calculate FFT`
6. Go to fitting tab, Add a Lorentzian and fit. (The fit will be bad)
7. Step to the next run. An error box appears `‘MuonBasePair’ object has no attribute ‘forward_group’`.

**Platform/Version (please complete the following information):**
Windows

AnthonyLim23, 2021-09-10:
This also happens in muon analysis:

Load data
Create a phase table
Create a phasequad (the + button on the phase tab)
Change run

AnthonyLim23, 2021-09-10:
This is coming from the corrections presenter (raise the error again instead of the warning message):
```
Traceback (most recent call last):
  File "C:\Users\BTR75544\work\dev\mantid\scripts\Muon\GUI\Common\thread_model.py", line 108, in warning
    self._exception_callback(message)
  File "C:\Users\BTR75544\work\dev\mantid\scripts\Muon\GUI\Common\corrections_tab_widget\corrections_presenter.py", line 136, in handle_thread_error
    self.view.warning_popup(error)
  File "C:\Users\BTR75544\work\dev\mantid\scripts\Muon\GUI\Common\corrections_tab_widget\corrections_view.py", line 76, in warning_popup
    warning(message, parent=self)
  File "C:\Users\BTR75544\work\dev\mantid\scripts\Muon\GUI\Common\message_box.py", line 11, in warning
    raise ValueError(error)
ValueError: 'MuonBasePair' object has no attribute 'forward_group'
```

DavidFair, 2021-09-10:
Fixed by #32464

PR #32464 (2021-09-10): Check pair instance is a MuonPair
**Description of work.**
This PR fixes a bug where it is assumed that the group/pair context only contains `MuonPair`s in the .pairs attribute. This is not true as a `MuonBasePair` is apparently created by the PhaseQuad tab.

The solution is to only perform the check if the pair is a `MuonPair`.

**To test:**
Muon Analysis
Load MUSR 62260
Create a phase table
Create a phasequad (the + button on the phase tab)
Step to the next run and wait
There should be no error message pop-up that appears.

Fixes #32455

*This does not require release notes* because **it was [... cut]
closed 2021-09-10 by DavidFair