mantidproject/mantid #32478: IDA fitting tabs do not use the first point when fitting

Harrietbrown, 2021-09-10:
In the Indirect Data Analysis fitting tabs the fit results do not cover the first point in the data even is startX is equal to the start value in the data. this can also happen with the last point as well.

AnthonyLim23, 2022-01-05:
@Harrietbrown do you think this worth fixing for the release?

Harrietbrown, 2022-01-05:
It is probably worth having a go at.

Harrietbrown, 2022-01-27:
This is a result of the IqtFit algorithm converting the workspace to a histogram before processing it. It is not clear why it does that but at first glance it does not seem nessarcery. I do not think it would be a good idea to put it in at this point in the release since we are not certain on if it is not making another issue. so I will move it inot the next milestone.

PR #33452 (2022-02-17): Remove change to histogram in IqtFit
**Description of work.**
This commit removes the conversion to histogram in IqtFit. It is not clear why this is done as all it ends up doing is stopping the fit from using the first value.
It also adds rounding to the ranges given to QENSFitSequential in IDA to stop the same issue in other tabs

**Report to:** @SpencerHowells

**To test:**

1. open Indirect Data Analysis Iqtfit tab
2. Load irs26176_graphite002_iqt spectra 0
3. set Exponentials to 1 and endX to 0.20
4. click run
5. show data on workspace `irs26176_graphite002_iqt_IQt_seq_1E_0_0_Workspace`
6. [... cut]
closed 2022-02-17 by gemmaguest