mantidproject/mantid #32521: MultiFile ILL data does not validate paths with spaces - Indirect Data Reduction

DanielMurphy22, 2021-09-16:
**Original reporter:** Mona Sarter

**Describe the bug**
For ILL data on the Indirect Data Reduction interface, multiple runs cannot be reduced if the file path includes a space.
This is always the case for data stored on One Drive, as was the case for this user.

**To Reproduce**
1. Ask @DanielMurphy22 For test data or have access to some ILL indirect data ( this data was from IN16B).
2. Put the data in a folder with a space in its name e.g. ``My Data Folder``
3. Open Mantid Workbench, set the facility to ILL and instrument to IN16B
4. Open the Interface Indirect Data Reduction 
5. On the ILL Energy Transfer tab, Browse for Sample Files. Shift click to select multiple files in the ``My Data Folder`` and click Open. Then at the bottom of this tab, click the Run button. 
6. You will get a notification "Please correct the following: Run File is invalid" and a warning message "Failed to validate indirect tab input!"

**Expected behavior**
ILL Indirect data can be reduced for multiple runs even when the path includes spaces. The validation works correctly for ISIS data on One Drive.

**Screenshots**
[image]

**Platform/Version (please complete the following information):**
 - OS: Found on Windows 10 but can reproduce on MacOS BigSur
 - Mantid Version Found on 6.1.0, but probably been around for a few releases.

SpencerHowells, 2021-09-16:
This has been around for a long time. I found it on a PC which has the default setup folder ‘My PC’

gvardany, 2021-09-16:
Indeed, this was like that since the beginning of time. This has to do with multiple file parsing, which fails if there is a space in the path.

closed 2023-01-15 by stale[bot]

darominski, 2023-01-16:
I believe this was fixed with #33780.