mantidproject/mantid #32534: Fix event filtering in LoadEventNexus

DannyHindson, 2021-09-17:
The filtering on start and end time using the FilterByTimeStart and FilterByTimeStop parameters isn't working when the input nexus file contains events that aren't already sorted by pulse time
eg
```
event_ws = Load(Filename=r'POLARIS00129560.nxs', FilterByTimeStart=308, FilterByTimeStop=309)
```
returns over 11 million events whereas it should be only 51098

Very similar problem to #32495 but same sorting code is replicated in LoadBankFromDiskTask::prepareEventId

DannyHindson, 2021-09-17:
This one's actually a bit tricky to fix. The LoadEventNexus algorithm currently assumes you can apply a pulse time filter at load time by adjusting the event id range that gets loaded (expressed as a start_event and stop_event range). This is only possible if the events are sorted on pulse_time - which isn't true in all Nexus event files.
So to provide the ability to filter using the FilterStartTime, FilterStopTime parameters, we need to load all the events and then filter down as a post process. This will be slower and perhaps removes the point of these parameters. You may as well just filter down using FilterByTime afterwards

martyngigg, 2021-09-22:
Fixed by #32547

closed 2021-09-22 by martyngigg