mantidproject/mantid #32578: Adding fitting function for muon results analysis

AnthonyLim23, 2021-09-24:
This is an epic for the main muon fitting functions that need adding for the model analysis.

AnthonyLim23, 2021-09-24:
General:
#32579
https://github.com/mantidproject/mantid/issues/32580
https://github.com/mantidproject/mantid/issues/32581
https://github.com/mantidproject/mantid/issues/32582

Magnetism
https://github.com/mantidproject/mantid/issues/32583
https://github.com/mantidproject/mantid/issues/32584
https://github.com/mantidproject/mantid/issues/32585
https://github.com/mantidproject/mantid/issues/32586

super conductivity
https://github.com/mantidproject/mantid/issues/32587
https://github.com/mantidproject/mantid/issues/32588

sf1919, 2022-01-04:
All the General and Magentism fitting functions have been added to v6.3. Details to enable work on super conductivity fitting functions have yet to provided - when asked the response was that more thought is required before proceeding. As we have fulfilled all the fitting functions we are able to at this time I have removed the milestone 6.3 and added the 'Awaiting User Response' label.

closed 2023-02-19 by stale[bot]