mantidproject/mantid #32605: Windows Fitting Smoke Tests

DanielMurphy22, 2021-09-27:
Before testing:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

Afterwards:
- Comment below with any issues you came across.
- If no issues were found, or they are now all resolved, please close the testing issue.
- Check the master issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :muscle: :runner:

* Fitting to data [directions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/01_fitting_models_to_data.html#fitting-models-to-data)
- [ ] Simple fit 
- [ ] Activating the fitting tools 
- [ ] Selecting the fitting range 
- [ ] Other Fit Function Settings properties 
- [ ] Setting up a fit model 
- [ ] Adjusting fit function parameter 
- [ ] Saving a model 
- [ ] Tying and constraining fit parameters 
- [ ] Execute your fit 
- [ ] Inspecting a fit result 
* Additional options [directions](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/03_fit_model_choices.html)
- [ ] Non peak model + background 
- [ ] User defined function

RichardWaiteSTFC, 2021-09-27:
Fit browser > Display > Quality option doesn't seem to plot anything...

RichardWaiteSTFC, 2021-09-27:
When adding a tie to a parameter it doesn't update the value of the parameter (but it does when you load the function from a  string). Indeed if it is sigma it can still be changed with the sliders.

1.  Add this function
`name=Gaussian,Height=487,PeakCentre=1,Sigma=2,constraints=(438.3<Height<535.7);ties=(f0.Sigma=f0.PeakCentre)`
Note sigma is set to same as PeakCentre = 1
2. Change Sigma with the sliders
3. Remove the tie, set Sigma != PeakCentre then add the tie to f0.PeakCentre again - the value isn't updated 

Could this be related to #32365

RichardWaiteSTFC, 2021-09-27:
If you plot a function by right clicking on a function in the browser (not using Display > Plot guess) - the Display > Remove Guess button is not enabled and when both an individual peak and the whole function guess is plotted removing guess curves  doesn't remove the single function curve (just the composite guess). You can remove it however by right-clicking again on the function and remove plot - but it might be good if Remove Guess removed all guess curves.

[image]

RichardWaiteSTFC, 2021-09-27:
Just a heads up this is still an issue #31660

DanielMurphy22, 2021-09-27:
@RichardWaiteSTFC are all these not release stopping? If so are you ready to close this and I'll sweep up the loose ends after the release.

RichardWaiteSTFC, 2021-09-27:
No all minor I think so I'll close!

closed 2021-09-27 by RichardWaiteSTFC