mantidproject/mantid #32607: Windows Plotting Smoke Tests

DanielMurphy22, 2021-09-27:
Before testing:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

Afterwards:
- Comment below with any issues you came across.
- If no issues were found, or they are now all resolved, please close the testing issue.
- Check the master issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :chart_with_upwards_trend: :bar_chart:

* 1D plotting:[instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/03_displaying_1D_data.html#displaying-1d-data)
 - [ ] Simple plot 
 - [ ] Another way to plot 
 - [ ] Adding curves to existing plots 
 - [ ] Also, test out [waterfall](https://docs.mantidproject.org/nightly/plotting/WaterfallPlotsHelp.html#waterfall-plots) and [tiled]( https://docs.mantidproject.org/nightly/plotting/1DPlotsHelp.html#tiled-plots)
 - [ ] Check Toolbar buttons

* 2D plotting: [instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/04_displaying_2D_data.html#displaying-2d-data)
 - [ ] Plot all spectra 
 - [ ] Change colour map 
 - [ ] [Contour plot](https://docs.mantidproject.org/nightly/plotting/ColorfillPlotsHelp.html#contour-plots) (under 3D menu) 
 - [ ] Check Toolbar buttons

* [3D plotting](https://docs.mantidproject.org/nightly/plotting/3DPlotsHelp.html):
 - Load some data eg `LOQ7041` from the ISIS sample data
 - [ ] 3D surface
 - [ ] 3D wire frame  
 - [ ] Check Toolbar buttons

 ## Sliceviewer
 - [ ] Overly long instructions (don't spend hours!) and data [here](https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html). In particular try editing the data in a workspace while it is open in Sliceviewer! ([step 10](https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html#alter-the-underlying-workspace))

SilkeSchomann, 2021-09-27:
SliceViewer does not close when the workspace gets deleted.

DanielMurphy22, 2021-09-27:
Apart from these 2 issues, @SilkeSchomann are you happy with everything else? (I'll make sure these 2 are raised)

SilkeSchomann, 2021-09-27:
Yes, apart from that it looks fine :-)

DanielMurphy22, 2021-09-27:
> SliceViewer does not close when the workspace gets deleted.

Issue raised here: https://github.com/mantidproject/mantid/issues/32638

closed 2021-09-27 by DanielMurphy22