mantidproject/mantid #32611: MacOS Clean/Dirty Install Smoke Tests

DanielMurphy22, 2021-09-27:
Before testing:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

Afterwards:
- Comment below with any issues you came across.
- If no issues were found, or they are now all resolved, please close the testing issue.
- Check the master issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :soap: :hankey:

### Dirty install

* Make sure that you have several versions of Mantid installed
 * Last release
 * A nightly
 * If possible an old release
* Install the latest version of the new Mantid
- [ ] Check that Mantid boots up correctly

### Clean install

* Remove all existing Mantid versions and associated files

**On Windows**:
 * Uninstall the program
 * Clear shortcuts from desktop
 * Clean out the registry
  * Load regedit (Command Prompt > regedit)

**On macOS** :
 * Remove the application
 * Remove the `~/.mantid directory`
 * Remove `~/Library/Preferences/org.mantidproject.MantidPlot.plist`

**On Linux** :
 * Remove the package: `/opt/Mantid`
 * Remove `~/.config/Mantid`
 * Remove `~/.mantid/`

* Re-install the latest version of the new Mantid
- [ ] Check that Mantid boots up correctly

DanielMurphy22, 2021-09-27:
Tested on BigSur, all works fine!

closed 2021-09-27 by DanielMurphy22

DavidFair, 2021-09-27:
Tested:
6.1->6.2 
5.1.1->6.2