mantidproject/mantid #32625: Ubuntu Clean/Dirty Install Smoke Tests

DanielMurphy22, 2021-09-27:
* Follow the instructions [here](https://download.mantidproject.org/ubuntu.html) to install the .deb for Ubuntu

Before testing:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

Afterwards:
- Comment below with any issues you came across.
- If no issues were found, or they are now all resolved, please close the testing issue.
- Check the master issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :soap: :hankey:

### Dirty install

* Make sure that you have several versions of Mantid installed
 * Last release
 * A nightly
 * If possible an old release
* Install the latest version of the new Mantid
- [ ] Check that Mantid boots up correctly

### Clean install

* Remove all existing Mantid versions and associated files

**On Windows**:
 * Uninstall the program
 * Clear shortcuts from desktop
 * Clean out the registry
  * Load regedit (Command Prompt > regedit)

**On macOS** :
 * Remove the application
 * Remove the `~/.mantid directory`
 * Remove `~/Library/Preferences/org.mantidproject.MantidPlot.plist`

**On Linux** :
 * Remove the package: `/opt/Mantid`
 * Remove `~/.config/Mantid`
 * Remove `~/.mantid/`

* Re-install the latest version of the new Mantid
- [ ] Check that Mantid boots up correctly

gemmaguest, 2021-09-27:
Mantid starts up fine in both cases. Just a couple of notes (I'm not sure if this is expected):
- the executable is not in my path
- with the dirty install I didn't get the About screen pop-up

DanielMurphy22, 2021-09-28:
@gemmaguest out of interest, did you install via the .deb?

gemmaguest, 2021-09-28:
> @gemmaguest out of interest, did you install via the .deb?

yes

DanielMurphy22, 2021-09-28:
> * with the dirty install I didn't get the About screen pop-up

Raised this issue: https://github.com/mantidproject/mantid/issues/32653

closed 2021-09-28 by DanielMurphy22