mantidproject/mantid #32627: Ubuntu Interface Smoke Tests

DanielMurphy22, 2021-09-27:
* Follow the instructions [here](https://download.mantidproject.org/ubuntu.html) to install the .deb for Ubuntu

Before testing:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

Afterwards:
- Comment below with any issues you came across.
- If no issues were found, or they are now all resolved, please close the testing issue.
- Check the master issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :facepalm:

## ISIS only, if possible, so catalogue access can be tested

- [ ] Mantid opens without errors or warnings 
- [ ] Every option in `Interface` opens a GUI 
- [ ] Load some test data 
- [ ] Access Catalogue through algorithms (use Facilities account, same as for IDAaaS): [CatalogLogin](https://docs.mantidproject.org/v3.9.0/algorithms/CatalogLogin-v1.html) and [CatalogGetDataFiles](https://docs.mantidproject.org/v3.9.0/algorithms/CatalogGetDataFiles-v1.html)
- [ ] Saving/loading projects works  [directions here](https://www.mantidproject.org/MantidPlot:_The_File_Menu#File-.3E_Save_Project_as...) 
 - [ ] Alter preferences in [File > Settings](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/useful/01_save_settings.html#settings) and check they are obeyed

gemmaguest, 2021-09-27:
The Indirect->Data Analysis GUI seems to have a lot of wasted space on the Elwin tab, and doesn't fit on my screen:

[image]

gemmaguest, 2021-09-27:
I think this is just that the instructions are out of date, but for project saving:
- there is no File->Save Nexus menu. However, Save->Save Nexus works.
- there are no print or export/import options

gemmaguest, 2021-09-27:
Just had a crash re-enabling grid lines in the Settings dialog. I think it happened when I was closing the settings dialog.

[image]

closed 2021-09-29 by martyngigg