mantidproject/mantid #32630: Ubuntu Workspace/Algorithm Smoke Tests

DanielMurphy22, 2021-09-27:
* Follow the instructions [here](https://download.mantidproject.org/ubuntu.html) to install the .deb for Ubuntu

Before testing:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

Afterwards:
- Comment below with any issues you came across.
- If no issues were found, or they are now all resolved, please close the testing issue.
- Check the master issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :man_astronaut: :woman_astronaut: :space_invader:

* Load some workspaces, try 5 from different instruments
* Show detectors, check Instrument Viewer, follow  directions [in the documentation](https://www.mantidproject.org/MantidPlot:_Instrument_View)
  - [ ] Render tab 
  - [ ] Draw tab 
  - [ ] Pick tab 
  - [ ] Instrument tree tab 
* Test 5 - 10 random algorithms, follow [the documentation](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/algorithms_workspaces_and_histories/02_algorithms.html#algorithms) and [pick your favourite](https://docs.mantidproject.org/v3.9.0/algorithms/)
 - [ ] Help button 
 - [ ] Validation of inputs 
 - [ ] Overwriting workspace 
 - [ ] Run the algorithm 
- [ ] Check right-click options of the workspaces, they are listed [in the documentation](https://www.mantidproject.org/MantidPlot:_The_Workspace_Menu#Save_Nexus).

gemmaguest, 2021-09-27:
Python control does not seem to create the instrument viewer window - if I understand correctly, `inst_win = getInstrument("wsname")` should open an instrument viewer window, but nothing happens

gemmaguest, 2021-09-27:
Clicking and dragging on the colorbar does not change the min/max values

closed 2021-09-29 by martyngigg