mantidproject/mantid #32635: Missing parentheses from print statements in example scripts

sf1919, 2021-09-27:
**Describe the outcome that is desired.**
In some of the sample scripts in the documentation there are no parentheses around the print statements. Current versions of python require this. It would be good to update the scripts to reflect this.

**Describe any solutions you are considering**
Updated script examples with print statements.

**Additional context**
Example from Min Algorithm - https://docs.mantidproject.org/v3.9.0/algorithms/Min-v1.html

Currently
```print "Minima for spectrum 0 is Y = ",minim.dataY(0)[0]," and it occurs at X between ", minim.dataX(0)[0], " and ", minim.dataX(0)[1]```

Revised version should be
```print ("Minima for spectrum 0 is Y = ",minim.dataY(0)[0]," and it occurs at X between ", minim.dataX(0)[0], " and ", minim.dataX(0)[1])```

martyngigg, 2021-09-27:
Ah. This is an interesting one. That link is for `v3.9.0` when we were still using Python 2 and we effectively freeze the docs for old versions. Was there a link that brought you to that page rather than https://docs.mantidproject.org/algorithms/Min-v1.html ?

sf1919, 2021-09-27:
I thought I'd found it via the search box. However I've just tried to repeat that and managed to get the correct version of the page with parantheses. Now I'm somewhat confused!

sf1919, 2021-09-27:
Ok so I think I've worked out how I did it. I followed the CatalogLogin link from here - https://github.com/mantidproject/mantid/issues/32606 which takes you to v3.9.0 and then searched from there.....

martyngigg, 2021-09-27:
Ah okay. We've just got an outdated link in those docs then I think so we can update that and I'll close this. Thanks!

closed 2021-09-27 by martyngigg