mantidproject/mantid #32640: Ubuntu Dirty install 6.2 release candidate

DanielMurphy22, 2021-09-27:
**Describe the bug**
A dirty install on Ubuntu seems to not show the Mantid About menu (first time startup)

https://github.com/mantidproject/mantid/issues/32625#issuecomment-927889232

martyngigg, 2021-09-28:
I can confirm that I do *not* see this on CentOS7 (IDAaaS).

DanielMurphy22, 2021-09-28:
@ConorMFinn and I have tested this a few different ways and I think I’m convinced that it’s working the way the code intends to, although it's a little odd. I think for @gemmaguest and myself, when we opened 6.2, the last version that had been opened must have been either a build or nightly after 6.1. The About widget checks versions by the Release notes URL (qthelp://org.mantidproject/doc/release/v6.2.0/index.html) and a nightly or build after 6.1 would have had the link to the 6.2 release notes. You could argue that we aren't quite following the "Do not show until next release" faithfully as a previous nightly counts as the new release in this case.  I can reproduce this behaviour on Windows as well as Ubuntu.

It may not be worth making a code change this late as:
- Opening and closing 6.1, then installing 6.2 and opening DOES show the About MantidWorkbench interface.
- I tried also opening a nightly 6.0.20210215.2130 and installing and opening 6.1 and the About MantidWorkbench interface did not show, so this "bug" has been around for at least one release, probably always.

DanielMurphy22, 2021-09-28:
Not worth fixing for this release, raised an in depth issue here: https://github.com/mantidproject/mantid/issues/32653

closed 2021-09-28 by DanielMurphy22