mantidproject/mantid #32647: Memory not released to the system

darominski, 2021-09-28:
While running some memory-intense reduction algorithms for PANTHER instrument at the ILL I have found that the memory consumption keeps increasing regardless of programmed cleanup of intermediate workspaces that take the bulk of the memory. it seems that the memory is indeed only partially released after a workspace is deleted. It is possible to reproduce this behaviour using a sample workspace, so the issue is not instrument- or algorithm-specific. This is related to another issue: #31777 .

**Describe the bug**

1. Memory is not properly released after a workspace is deleted. Only a fraction of the memory consumed is released. To release memory, closing of Mantid altogether is necessary.
2. Small issue: The reported size of the workspace in the information available can be extremely far from reality, see the proposed way to reproduce this issue below.

**To Reproduce**

In the workbench, create a workspace consuming a large amount of memory, for example to allocate about 4 GB run the following algorithm:
```
CreateSampleWorkspace(
    OutputWorkspace='memory_test',
    WorkspaceType='Histogram',
    XUnit='TOF',
    NumEvents=100000,
    NumBanks=100,
    BankPixelWidth=200
)
```
Deleting this workspace frees about 1.1 GB on the tested machine. The remaining 2.9 GB is still not accessible from outside of this instance of Mantid workbench, but it seems like it can be re-allocated, at least partially, within the same instance.

Second issue:  The reported memory used is 9155 MB, which is about double of the actual memory consumed. Sometimes this value is given as 178 883 GB, which is not physically possible on the machine used for testing.

**Expected behavior**

Memory is properly released after a workspace is deleted (high priority) and the memory used is reported correctly (low priority).

**Platform/Version (please complete the following information):**
 - OS: Ubuntu
 - OS Version: 18.04
 - Mantid Version 6.2 

The issue exists most probably on other operating systems as well.

**Additional context**

martyngigg, 2021-09-28:
@darominski Thanks for raising this. Can I just check you are running workbench through the launch script: `launch_mantidworkbench.sh`? This should use `jemalloc`.

martyngigg, 2021-09-28:
I've just tried to reproduce this on CentOS 7 with the release candidate for 6.2 and it seems to release the memory as I expect but only if I do use the startup script.

darominski, 2021-09-28:
> @darominski Thanks for raising this. Can I just check you are running workbench through the launch script: `launch_mantidworkbench.sh`? This should use `jemalloc`.

Ah! That's it, when I use it, the memory release works fine. I was somehow not aware of the startup script. In this case we do not have an issue.

closed 2021-09-28 by darominski

martyngigg, 2021-09-28:
> Ah! That's it, when I use it, the memory release works fine. I was somehow not aware of the startup script. In this case we do not have an issue.

Thanks for checking. It is a bit annoying that we need this extra script but the `jemalloc` allocator has to be set to load before the process runs so it can't just be included inside the usual setuptools-generated startup script.