mantidproject/mantid #32679: FindPeaks crashes if run on histograms containing a single bin

DannyHindson, 2021-09-30:
This is clearly not a realistic histogram to try finding peaks in - the reason I noticed this is that I had a workspace containing many single bin histograms and I had forgotten to transpose it before running FindPeaks. It shouldn't crash Mantid though...

DannyHindson, 2021-09-30:
Happens if FWHM=7 which gives npts=5 and halfwidth=2. This code crashes in SmoothData.cpp:
```
/// Returns a smoothed version of histogram. npts must be odd.
Histogram smooth(const Histogram &histogram, int npts) {
  if (npts == 0)
    return histogram;
  int halfWidth = (npts - 1) / 2;

  Histogram smoothed(histogram);

  const auto &Y = histogram.y();
  const auto &E = histogram.e();
  auto &newY = smoothed.mutableY();
  auto &newE = smoothed.mutableE();
  // Use total to help hold our moving average
  double total = 0.0, totalE = 0.0;
  // First push the values ahead of the current point onto total
  for (int k = 0; k < halfWidth; ++k) {
    if (Y[k] == Y[k])
      total += Y[k]; // Exclude if NaN
    totalE += E[k] * E[k];
  }
```

DannyHindson, 2022-07-26:
This script will recreate the crash. The workspace sort of contains a peak but only when you do "Plot Bins". This data is not formed correctly for the FindPeaks algorithm because the data needs to be in a single histogram. This could be achieved by running the Transpose algorithm on the ws workspace. The script shouldn't cause a crash though and it's fixing the crash that's the subject of this issue (probably by adding a validation somewhere that returns an error message):
```
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

ws = CreateWorkspace(OutputWorkspace='test', DataX='1', DataY='0,0,0,2,5,2,0,0,0,0', NSpec=10)

peaks_ws = FindPeaks(ws)
```

robertapplin, 2024-05-13:
Fixed by https://github.com/mantidproject/mantid/pull/37263

PR #37263 (2024-05-10): 32679 fixed FindPeaks crash when the number of bins are not sufficient
### Description of work
Fixed a crash on `FindPeaks` algorithm when run on histograms containing a single bin by raising an error message on this condition.

Fixes #32679. 

#### Further detail of work

### To test:
1. Run the below code and the algorithm should not crash instead should throw a meaningful error
```
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

ws = CreateWorkspace(OutputWorkspace='test', DataX='1', DataY='0,0,0,2,5,2,0,0,0 [... cut]
closed 2024-05-13 by robertapplin