mantidproject/mantid #32892: Support Sphere in preset shapes supplied to SetSample

RichardWaiteSTFC, 2021-11-03:
**Describe the solution you'd like**
At present there are some shapes (e.g. hollow cylinder) that are preset and can be created in `SetSample `like so
```
ws = CreateSampleWorkspace()
# Use geometry from environment but set different height for sample
SetSample(ws, Geometry={'Shape': 'HollowCylinder', 'Height': 4.0, 'InnerRadius': 0.8, 'OuterRadius': 1.0, 'Center': [0.,0.,0.]},
    Material={'ChemicalFormula': 'V0.95-Nb0.05', 'SampleNumberDensity': 0.0119})
```
However to make a sphere we need to provide an xml string
```
shape= '''<sphere id="V-sphere">
<centre x="0.0"  y="0.0" z="0.0" />
<radius val="0.0025"/>
</sphere>'''
CreateSampleShape('SXD33338_sm_deadCor', shape)
SetSampleMaterial('SXD33338_sm_deadCor', SampleNumberDensity=0.0119, 
    ScatteringXSection = 5.197,AttenuationXSection=4.739,ChemicalFormula='V0.95 Nb0.05')
```
The solution I would like would be to support spherical shapes as so
```
SetSample('SXD33338_sm_deadCor', Geometry={'Shape': 'Sphere', 'Radius': 0.0025},
          Material={'ChemicalFormula': 'V0.95-Nb0.05', 'SampleNumberDensity': 0.0119}) 
```

**Describe alternatives you've considered**
Alternatively would it be possible to provide a better error to inform the user the shape provided is invalid.
Running this code 
```
SetSample('SXD33338_sm_deadCor', Geometry={'Shape': 'Sphere', 'Radius': 0.0025},
          Material={'ChemicalFormula': 'V0.95-Nb0.05', 'SampleNumberDensity': 0.0119}) 
```
Throws
```
Invalid value for Geometry: For Sphere shape Height is required
RuntimeError: SetSample-v1: Some invalid Properties found: [ Geometry ]
```
It would be better to say 'sphere' is not supported

**Additional context**

DannyHindson, 2021-11-23:
check radius > 0

DannyHindson, 2021-11-23:
David seemed to get quite close to adding this a few years ago in this PR but don't think it was ever merged: #20237.

gvardany, 2022-01-04:
Sphere should be supported with Radius of course and not height. I think the misleading error just comes from the fact, that for all the existing shapes (flat, cylinder, annulus) height is the common mandatory parameter.

PR #34264 (2022-08-10): Support sphere in SetSample
**Description of work.**
Previously anyone wanting to use a sphere in  SetSample would have to do so using an xml string. This PR adds sphere to the preset shapes available.

**To test:**

1. Build Mantid
2. Run the following code and it shouldn't crash
```
ws = CreateSampleWorkspace()
SetSample(ws,
        Geometry={'Shape': 'Sphere',
                  'Radius': 2.0, 'Center': [0.,0.,0.]})
```
3. Code review

Fixes #32892 

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

 [... cut]
closed 2022-08-10 by martyngigg