mantidproject/mantid #32894: Add QLwat program to BayesQuasi

StephenSmith25, 2021-11-03:
**Is your feature request related to a problem? Please describe.**

A new fitting module has been added to the BayesQuasi fortran code by Spencer. The code is now hosted here:

https://github.com/mantidproject/bayesfitting/

The new code is added in https://github.com/mantidproject/bayesfitting/pull/2

**Describe the solution you'd like**

Once this code is added we should be able to add something like:

```
  elif prog == 'QLw':
                workflow_prog.report('Processing Sample number {0} as Stretched Exp'.format(spectrum))
                nd, xout, yout, eout, yfit, yprob = QLwat.qlwat(numb, Xv, Yv, Ev, reals, fitOp,
                                                                Xdat, Xb, Yb, Wy, We, dtn, xsc,
                                                                wrks, wrkr, lwrk)

```

If we are not fully switching over to the PIP version of the library, then this new additional program will have to be optional. 

We'll need to update the GUI as well.

thomashampson, 2022-03-08:
@AnthonyLim23 will create an issue elsewhere to cover this

AnthonyLim23, 2022-03-10:
I am reopening this. Reading it closely, it is to update the GUI and scripts in Mantid so that the new option can be used in Mantid. This is to be done after the code has been added to quasibayes repo

closed 2023-03-17 by stale[bot]