mantidproject/mantid #32895: Muon analysis plot toolbar not compatible with matplotlib>=3.3

StephenSmith25, 2021-11-03:
**Describe the outcome that is desired.**

**The toolbar should work with newer versions of matplotlib.**

**The issue**

Matplotlib implementation changed in 3.3 so the self._active variable no longer exists. 

- To see the issue install matplotlib>3.3 and try to use the Muon toolbar zoom feature.

https://github.com/mantidproject/mantid/blob/4b3013731d1f2a77a18634d515f6d7f6045fbc96/qt/python/mantidqtinterfaces/mantidqtinterfaces/Muon/GUI/Common/plot_widget/plotting_canvas/plot_toolbar.py#L82

Ideally, we should refrain from using private matplotlib data members, as they seem to change their private implementations frequently.

AnthonyLim23, 2021-11-03:
what is the "new equivalent" and when do we plan to update the matplotlib version in Mantid?

StephenSmith25, 2021-11-03:
I guess there's strictly no equivalent to this as it wasn't using public api from Matplotlib. The update to the matplotlib version will likely be when we fully switch to conda.

AnthonyLim23, 2021-11-03:
we just need it to know when to update the quickedit widget. 

What is the expected date/release for the move to conda?

AnthonyLim23, 2022-04-06:
@martyngigg as far as I am aware there is no equivalent for this in the newer version of matplotlib. Should we just let it "expire"?

martyngigg, 2022-05-03:
@AnthonyLim23 Sorry, I somehow missed the notification of your last message.

I don't think we can leave this as we have to move the matplotlib version forward so we need to find a way to solve it or the GUI will stop working. I'll leave this in the 6.4 milestone as we will need to solve it for that.

martyngigg, 2022-05-03:
My suggestion here would be that, as we have pretty much overidden all of the base behaviour, that we define a variable internally ourselves that we set appropriately in `zoom` & `pan`

PR #33813 (2022-05-05): Update muon plots for new matplotlib
**Description of work.**

Matplotlib implementation changed in 3.3 so the self._active variable no longer exists. This causes the muon plotting zoom and pan buttons to no longer work. This fixes that.

**To test:**

Open muon analysis
Load some data (which is irrelevant)
Press the zoom button on the toolbar
Check that autoscale is unchecked and unselectable
select a zoom region on the plot
Press the zoom button
The x and y values under stated under the plot should match what is shown
The autoscale checkbox should be selectable

Repeat the above, but press p [... cut]
closed 2022-05-05 by martyngigg