mantidproject/mantid #32909: LoadAndMerge ignores loader version

darominski, 2021-11-05:
This bug has been discovered while working on PR #32155 . Since that PR may not go forward in the nearest future, a separate issue is created for this particular bug. 

**Describe the bug**

`LoadAndMerge` does not take into account the user's choice for the loader version to be called. It always uses the one with highest confidence, and in case confidence is equal: the first one in alphabetical order. This happens even though `LoadAndMerge` verifies if the requested version actually exists.

**To Reproduce**

It seems that outside PR #32155 , the only loader currently having two versions is `LoadMuonNexus`.
1. In workbench, set log level to debug.
2. Call `LoadAndMerge` on some data that could be loaded by LoadMuonNexus, e.g. argus0026287.nxs available in the Data/UnitTest directory, make sure that `LoaderVersion` is set to 1.
3. There should be an error, as version 1 is not able to load this data file (confirm by calling v1 of `LoadMuonNexus`, e.g. using a script). In reality, there is no error.
4. Verify in the logs, which version of the `LoadMuonNexus` has been used by `LoadAndMerge`.
5. Repeat with setting `LoaderVersion` set this time to 2, the behaviour should not change.

**Expected behavior**

LoadAndMerge respects user's choice of the LoaderVersion.

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

Found on Ubuntu 18.04, but I believe this issue is platform-agnostic.

**Additional context**

AnthonyLim23, 2021-11-05:
I think the issue is that the data. If you try to load ARGUS 26287 into the muon GUI, then you get a warning message about the data not existing. If you look in the data archive itself a lot of the ARGUS runs are missing for the early cycles. 

If you use some data that is present it works much better:
```
out = LoadAndMerge(Filename='MUSR62260-62261,MUSR62255+MUSR62258')
print('out is a WorkspaceGroup, containing {0} workspaces'.format(out.getNumberOfEntries()))
print('the first item is the merged output of the runs 170257 and 170258 with the name {0}'.format(out.getItem(0).getName()))
print('the second item is the merged output of the runs 170300 and 170302 with the name {0}'.format(out.getItem(1).getName()))
```

darominski, 2021-11-05:
> I think the issue is that the data. If you try to load ARGUS 26287 into the muon GUI, then you get a warning message about the data not existing. If you look in the data archive itself a lot of the [...]

The issue has been spotted when we started developing a v2 of `LoadILLSANS` algorithm and realised that `LoadAndMerge` used as intermediary to call `LoadILLSANS` ignores our choice for the LoaderVersion. Unfortunately, I cannot give the original issue as the example as that PR is currently suspended, but it seems the issue is not specific to our new loader version and I used `LoadMuonNexus` as the only known to me example of a loader having currently two versions available.  

I don't seem to be able to find the data you have mentioned in your example, it does not seem to be a part of the unit/doc/system test data package nor the UsageExamples. Could you point me to where I could get them to test this out?

AnthonyLim23, 2021-11-05:
the data I used was just from the ISIS data archive. I can dig out some from the publicly  available data

AnthonyLim23, 2021-11-05:
```out = LoadAndMerge(Filename='EMU20882-20883,EMU20885+20886')
print('out is a WorkspaceGroup, containing {0} workspaces'.format(out.getNumberOfEntries()))
print('the first item is the merged output of the runs 170257 and 170258 with the name {0}'.format(out.getItem(0).getName()))
print('the second item is the merged output of the runs 170300 and 170302 with the name {0}'.format(out.getItem(1).getName()))
```

You can get the data from the Muon data set on the download page. It is in the folder called "cupper" (I think it is supposed to say copper)

darominski, 2021-11-05:
> ```
> print('out is a WorkspaceGroup, containing {0} workspaces'.format(out.getNumberOfEntries()))
> print('the first item is the merged output of the runs 170257 and 170258 with the name [...]

Thank you, this is then a better example, as both v1 and v2 of `LoadNexusMuon` can load data in the 'muon_cupper' directory. Nonetheless, the reason I am raising this issue is still there: if one tries to load a NXS Muon file in the 'muon_cupper' directory e.g. `EMU00020923.nxs` via `LoadAndMerge`, `LoadAndMerge` will ignore user's choice for the `LoaderVersion` and will always pick the v2.

PR #32910 (2021-12-06): LoadAndMerge fix for ignored loader version
**Description of work.**

This PR fixes the issue of LoadAndMerge algorithm ignoring user's choice of the loader version. The issue was passing arguments not by keyword but by their position.

**To test:**

Repeat the test from the issue #32909, this time when LoaderVersion is set to 1, there should be a clear error when an attempt is made to load the suggested file.

Fixes #32909. 

Release note added.

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### [... cut]
closed 2021-12-06 by gemmaguest